jperkin/node-rpio #1: Segmentation fault on binding require

Aldaviva, 2015-03-14:
### Environment ###
- Raspberry Pi Model B Rev. 2 (hardware: BCM2708, revision: 000d)
- Raspbian Wheezy 7.8
- Node 0.10.24
- rpio 0.4.2

### Steps to reproduce ###
1. `npm install rpio@0.4.2`
2. The `rpio.node` build succeeds.
3. `root@host# node`
4. `> require('rpio')`

### Result ###
```
Segmentation fault
```

The segfault is happening at the top of `rpio.js` while loading the native C binding:
```javascript
var binding = require('../build/Release/rpio');
```

I don't know how to debug this any further, but I'd be happy to try to help figure this out.

jperkin, 2015-03-14:
Hmm, thanks.  I dug out my older Raspberry Pi Model B (have been primarily developing on a B+) and was able to reproduce this.  Not sure what's causing it yet but will take a closer look, it does seem to be related to running on the older models.

jperkin, 2015-03-14:
Ok, try `rpio@0.4.3`, seems to work for me.

Aldaviva, 2015-03-15:
Thanks! It works now.

Your fix will be much appreciated whenever a build fails and a spinning red light turns on in the office.

closed 2015-03-15 by Aldaviva

jperkin, 2015-03-15:
Sweet ;)