jperkin/node-rpio #2: PWM units

jurafa, 2015-04-23:
Hi, I belieave there is an error in units in PWM in documentation. If refresh rate is set to 2048, than the frequecy is 9.375Hz and not 9.375kHz.

jperkin, 2015-04-23:
I'm pretty sure 19.2MHz / 2048 = 9.375kHz ;)

closed 2015-04-23 by jperkin

jurafa, 2015-04-23:
This is what my oscilloscope sais to it... :-)

[image]

jperkin, 2015-04-23:
Surely that's dependent on the pulse width you have chosen rather than the base frequency?

jperkin, 2015-04-23:
However, looking at this again I notice the doc is wrong in saying that 2048 is the minimum rate, as it's 4096 (4.6875kHz), so I've fixed that and published 0.4.4

jurafa, 2015-04-23:
This is my code. I measure data on UniPi board and its Analog output. I will try to meassure it directly on pin.

Raspberry Pi Details:
  Type: Model B+, Revision: 1.2

```javascript
var rpio = require('rpio');

var pin = 18;
var range = 1024;
var clockdiv = 2048;

rpio.setFunction(pin, rpio.PWM);
rpio.pwmSetClockDivider(clockdiv);
rpio.pwmSetRange(pin, range);

var data = 500;
rpio.pwmSetData(pin, data);}
```

jurafa, 2015-04-23:
Measured directly on the pin...

[image]

jperkin, 2015-04-23:
Looks pretty accurate to me?

* You start with a base frequency of 19.2MHz
* That is then divided by your clock divisor (2048) to produce a clock frequency of 9.375kHz
* You then choose a pulse width of 1024, which is used to divide up the chosen frequency.

```
19.2 * 1000000 / 2048 / 1024 = 9.155Hz
```

That's not exactly 9.149Hz as reported by your oscilloscope, but I'd guess that's either just rounding errors, or caused by your data value of 500 creating an uneven pulse - maybe changing it to 512 and a regular pulse will make it show 9.155?

Either way, I still don't see a bug here, though I will happily admit to being far from an expert in this field.

jurafa, 2015-04-23:
Ok. You are right. I did not fully understand the meaning of the range. I just learned something thanks to you. Thank's for being patient with me. When I saw the simple math how you calculate the frequency it all became clear. You should maybe cosider to add it to the documentation.

Thank's again.

jperkin, 2015-04-23:
Yeh I was thinking the same thing, I'll try to add something soon. Thanks!