jperkin/node-rpio #3: Unknown PWM channel on Pi Model B

mukulseth, 2015-06-06:
Hi,

I've been trying to run the PWM example from the repo on the RPi Model B and I keep getting the following error --

Unknown PWM channel for pin16 

using mode:physical - I'm sure the pin is correct as I can toggle the value between HIGH/LOW with basic write statements
using package version 0.4.4

RobQuistNL, 2015-12-04:
Same issue here... Using pigpio i can write to the pins (pigs p 17 |0 - 255| ) and do the same for pin 22 and 24, but when i open them here with GPIO pins i get the message "Unkonwn PWN channel" as well

jperkin, 2015-12-14:
Only certain pins support PWM.  On 26-pin variants you can only use GPIO18/P12, whereas on the 40-pin variants you can use GPIO12/Pin32, GPIO13/Pin33, GPIO18/Pin12, or GPIO19/Pin35.  There is a sentence to this effect in the README above the example, perhaps one of you can suggest better wording for either the README or the error message to make this more obvious?  Thanks.

RobQuistNL, 2015-12-19:
Well, the weird thing is that if I use other tools, such as pigpio, it does work as a PWM pin. Or am I missing something? I mean, i can use pigpio to write 0 to 255 to a pin and the MOSFET on that pin works accordingly (powering my ledstrip)

jperkin, 2015-12-19:
I'm not familiar with that software, but my guess (really the only explanation) is that it's simply doing software PWM, i.e. manually toggling the pins itself to generate PWM output.  My module doesn't support that, at least not natively (you can always implement it yourself with a suitable for loop), but does support hardware PWM using the pins specified, which is much more efficient.

jperkin, 2015-12-19:
I've added a paragraph attempting to explain this in the README.

RobQuistNL, 2015-12-21:
Ah right. In that case I guess this can be closed, right? Maybe add a nice to have feature that supports software PWM (didn't even know that was possible, thanks! :+1: ).

Thanks for the quick replies! :)

jperkin, 2016-01-08:
I'll close this for now, but adding software PWM seems like something I could add support for at some point.  Cheers.

closed 2016-01-08 by jperkin

jaqua, 2018-10-14:
@RobQuistNL How did you solve your pwm-issue? I'm also trying to power a led strip via software pwm...

RobQuistNL, 2018-10-14:
I think I used middleware and used pigpio, but its been a long time, don't really recall - Sorry!

eltoroit, 2021-02-28:
Any update on software PWM?