jperkin/node-rpio #5: Unable to npm install

nkolban, 2015-12-08:
I am trying to npm install the package but when I do, I get errors.   Attached is the 
[npm-debug.txt](https://github.com/jperkin/node-rpio/files/54808/npm-debug.txt)

jperkin, 2015-12-08:
Unfortunately I do not support the newer node.js releases at this time.  Everything up to 0.12 should be supported well, but since io.js started to break the ABI frequently it's difficult to keep up.  At some point I will have to move to NaN to have any chance of keeping up, which is unfortunate as it forces me back to C++, but I haven't had a few spare evenings to do this yet.

jperkin, 2015-12-18:
This is now supported as of v0.5.0.  For nodejs v4 and v5 you will need to ensure you install the GCC 4.8 packages, earlier GCC releases do not support C++11 required by newer v8.

closed 2015-12-18 by jperkin