jperkin/node-rpio #12: Access PWM Timings

infusion, 2016-05-04:
Hi,

is it possible to access the raw timings of the PWM with node-rpio? At the moment I set up a test with the python library rpio, which works perfectly well using the set_servo method: http://pythonhosted.org/RPIO/pwm_py.html

However, most of my code base is on JavaScript and I don't want to call python for this job. 

I haven't looked into your source yet. Do you use DMA for the PWM signal, too?

Thanks!

jperkin, 2016-05-04:
Yes, this library provides direct hardware access to set up the PWM exactly as you need. If the README doesn't make this clear could you give me an idea of how I can improve it?  I don't provide any convenience functions like the python module you linked to does, so you need to setup your clock/range/width values yourself.

infusion, 2016-05-05:
Okay, I really have to say thanks for this lib. I worked a little more with it under different scenarios and it is great!

Even if I now understand how the parameters are chosen for the methods you provide, it took some experimenting to get what it means. I was wondering what `pwmSetRange` means. You state "This determines the maximum pulse width.", so I was wondering how you change the pulse width. But it's just the scale of the data coming in via `pwmSetData`. Maybe a small note what happens when the data is outside the interval `[0, range]` (clamped, undefined behavior, ...) and if the range must be a power of two would be nice as well.

The last handy thing in the docs would be an equation to calculate a given duty cycle of let's say 1500us into a correct value within the range (A convenience function doing this is indeed not needed). But if you want to control a servo, you get the timing out of the specs and being able to calculate this quickly and correctly saves a lot of time. Maybe adding an example for a servo to the examples folder, too.

I also wrote a little snippet to handle a hall-effect sensor. You're free to add this to your examples:

```
var rpio = require('rpio');

var PIN_HALL = 7

rpio.open(PIN_HALL, rpio.INPUT, rpio.PULL_DOWN);

rpio.poll(PIN_HALL, function() {

        console.log("pulsed");

}, rpio.POLL_HIGH);
```

infusion, 2016-05-05:
Ah and one last question: A typical PWM signal for RC servos repeats every 20ms, resulting in 50Hz and therefore 19.2MHz*1000*1000/50Hz=384000, which is not a power of two. How can you change the period to 20ms then?

misterikkit, 2017-02-27:
A clock divider of 128 with a range of 3000 gives a 50Hz cycle, and a value of 150 would give a 1ms pulse. There's a pretty good explanation of how the numbers work here: http://raspberrypi.stackexchange.com/questions/4906/control-hardware-pwm-frequency/9725#9725

From my other reading, I thing you should be able to use values other than powers of two for the clock divider. I'll file a separate issue.

closed 2018-11-29 by jperkin