jperkin/node-rpio #13: poll forces IRQ to be disabled

infusion, 2016-05-07:
Hi,

running the example `poll()` snippet from #12 works fine, but when starting the program for the first time, I get a kernel message which produces the following `dmesg` output:

```
[   50.439526] irq 73: nobody cared (try booting with the "irqpoll" option)
[   50.439571] CPU: 0 PID: 988 Comm: node Not tainted 4.4.8+ #881
[   50.439587] Hardware name: BCM2708
[   50.439661] [<c0016c9c>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[   50.439702] [<c0013c20>] (show_stack) from [<c02e2e1c>] (dump_stack+0x20/0x28)
[   50.439749] [<c02e2e1c>] (dump_stack) from [<c005ced8>] (__report_bad_irq+0x34/0xfc)
[   50.439783] [<c005ced8>] (__report_bad_irq) from [<c005d25c>] (note_interrupt+0x244/0x2a4)
[   50.439815] [<c005d25c>] (note_interrupt) from [<c005ab34>] (handle_irq_event_percpu+0x158/0x1f0)
[   50.439848] [<c005ab34>] (handle_irq_event_percpu) from [<c005ac04>] (handle_irq_event+0x38/0x4c)
[   50.439880] [<c005ac04>] (handle_irq_event) from [<c005d990>] (handle_level_irq+0x9c/0x134)
[   50.439910] [<c005d990>] (handle_level_irq) from [<c005a0e4>] (generic_handle_irq+0x30/0x44)
[   50.439939] [<c005a0e4>] (generic_handle_irq) from [<c005a37c>] (__handle_domain_irq+0x58/0xb8)
[   50.439972] [<c005a37c>] (__handle_domain_irq) from [<c0010748>] (handle_IRQ+0x2c/0x30)
[   50.440003] [<c0010748>] (handle_IRQ) from [<c000940c>] (bcm2835_handle_irq+0x3c/0x58)
[   50.440036] [<c000940c>] (bcm2835_handle_irq) from [<c0571d64>] (__irq_svc+0x44/0x5c)
[   50.440053] Exception stack(0xd68bbeb0 to 0xd68bbef8)
[   50.440074] bea0:                                     00000000 0000000a c0887a40 c081cf74
[   50.440101] bec0: 00000000 c085e3e4 00000002 00000000 00000000 b4be53d4 00df36c0 d68bbf44
[   50.440122] bee0: d68bbf00 d68bbf00 c0025264 c0025268 60000113 ffffffff
[   50.440161] [<c0571d64>] (__irq_svc) from [<c0025268>] (__do_softirq+0xb4/0x314)
[   50.440195] [<c0025268>] (__do_softirq) from [<c002581c>] (irq_exit+0xbc/0x11c)
[   50.440229] [<c002581c>] (irq_exit) from [<c005a384>] (__handle_domain_irq+0x60/0xb8)
[   50.440262] [<c005a384>] (__handle_domain_irq) from [<c0010748>] (handle_IRQ+0x2c/0x30)
[   50.440290] [<c0010748>] (handle_IRQ) from [<c000940c>] (bcm2835_handle_irq+0x3c/0x58)
[   50.440318] [<c000940c>] (bcm2835_handle_irq) from [<c0572034>] (__irq_usr+0x54/0x80)
[   50.440335] Exception stack(0xd68bbfb0 to 0xd68bbff8)
[   50.440355] bfa0:                                     b3d0004c 00000010 00000010 00000000
[   50.440379] bfc0: be8b3708 be8b3738 be8b37a8 b4be72c8 00b8c46d b4be53d4 00df36c0 be8b3704
[   50.440401] bfe0: b4bf99d0 be8b36e0 b4be7504 b4be7504 60000010 ffffffff
[   50.440412] handlers:
[   50.440437] [<c030c684>] bcm2835_gpio_irq_handler
[   50.440457] Disabling IRQ #73
[  109.789711] gpiomem-bcm2835 20200000.gpiomem: gpiomem device opened.
```

Is it possible to circumvent this problem or do I have to add the kernel option `irqpoll`? I read the manual polling is quite slow compared to real IRQ's, or is this impossible with gpiomem?

Thanks Robert

scramble45, 2017-01-04:
I am also seeing this...

nueverest, 2017-05-07:
Sometimes I see IRQ 79, but my app keeps on running fine.

closed 2019-10-17 by jperkin

infusion, 2019-10-20:
Nice, thanks! :)