jperkin/node-rpio #14: Does not work on raspberry pi 3

kaalpurush, 2016-06-16:
When run on raspberry pi 3 latest raspbian image throws this error:

Error: /home/pi/rpio/node_modules/rpio/build/Release/rpio.node: invalid ELF header

/home/pi/rpio/node_modules/rpio/node_modules/bindings/bindings.js:83
        throw e

jperkin, 2016-06-16:
I don't yet have a Raspberry Pi 3 to test on unfortunately.  Could you do a clean build/install and send me the full output so I can see if there are any clues during the build?  Thanks.

sonnny, 2016-06-20:
It works on my rasbperry pi 3, I'm running Jessie lite, I just tried pwm-led.js example.
uname -a
Linux raspberrypi 4.4.9-v7+ #884 SMP Fri May 6 17:28:58 BST 2016 armv7l GNU/Linux
node -v
v4.2.1

kaalpurush, 2016-06-24:
I did perform a clean npm install. It's still showing the error.

uname -a
Linux raspberrypi 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux
node -v
v5.9.1
npm -v
2.15.6

```
/home/pi/rpio/node_modules/rpio/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: /home/pi/rpio/node_modules/rpio/build/Release/rpio.node: invalid ELF head                                                                                er
    at Error (native)
    at Object.Module._extensions..node (module.js:440:18)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
    at bindings (/home/pi/rpio/node_modules/rpio/node_modules/bindings/bindings.                                                                                js:76:44)
    at Object.<anonymous> (/home/pi/rpio/node_modules/rpio/lib/rpio.js:17:34)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
```

kaalpurush, 2016-06-24:
I had to rebuild with a reinstalled node-gyp and now everything works! Either my node-gyp was broken or was compiling for wrong processor architecture.

Thanks all for the help. You guys rock.

closed 2016-06-24 by kaalpurush

thisispete, 2018-06-01:
having the same problem, how do you re-build?

dauledk, 2018-12-18:
can we maybe reopen this issue, until we get a solution? I am running into the same issue.

jperkin, 2018-12-18:
I'd need some way to reproduce this.  I've never seen this issue, it's never appeared in the continuous integration tests, and it's also not something I would have any control over.  The only thing I can (and would be willing to) do is to add some notes to the documentation once we figure out the root cause.

dauledk, 2018-12-18:
@jperkin sounds fair enough to me. I'm currently trying to investigate the issue. My theory so far is that the compilation step is run against a wrong architecture (like @kaalpurush gave a hint about). The best thing I could envision for this package was not only to mention a possible gotcha in the documentation, but maybe even an improved error message. I will return when I have found out more 😄

jperkin, 2018-12-18:
The error is coming from the bindings module, and perhaps even node itself, so unfortunately there's no way I can improve that side of things.