jperkin/node-rpio #15: Using more than one SPI device

briandorey, 2016-07-05:
I am not able to read/write to more than one device using the library, I am using rpio.spiChipSelect(0); within the function to access the first spi device and then rpio.spiChipSelect(1); to access the second chip but this dosnt seem to work. 

Do I need to create separate instances of the rpio object for each SPI device?

gertvermeersch, 2016-07-05:
Hey Brian,
I'm only using one SPI device but the chipselect doesn't seem to work with me as well. I got it working by addressing the CSN pin manually using the rpio.write method instead before and after communicating over SPI.

briandorey, 2016-07-06:
gertvermeersch, thank you for the suggestion with the rpio.write, i will try it later today.

jperkin, 2018-11-29:
Haven't heard anything back on this one for a while so going to close it out.

closed 2018-11-29 by jperkin