jperkin/node-rpio #16: Cannot get switch events (press/release)

madmanblues, 2016-08-22:
I followed the readme instructions to read the status of a switch but there is no output on the console when I press or release the button.

I've installed rpio via npm
I added pi and root users to gpio group
I added the rule as root 
I connected the switch between PIN 6 and PIN 15 (GPIO022)
I ran the program as pi as well as root 

I never get an output when pressing the button

I logged to the console the rpio object and I got this (it may be useful):
{ domain: null, _events: {}, _maxListeners: 10 }

Can you please help me to start using this library?

jholster, 2016-09-04:
Do you pull up the pin 15? E.g. rpio.open(15, rpio.INPUT, rpio.PULL_UP)

jperkin, 2018-11-29:
Haven't heard back on this one for a while so going to close it out, please reopen if this is still an issue.

closed 2018-11-29 by jperkin