jperkin/node-rpio #18: usleep

ibmua, 2016-09-01:
It seems like microsecond sleep is seriously lacking precision. I was able to get much better results, though, still not too good, using node "microsleep" module to measure time and a silly while loop

```
while
	(
		microtime.now()
	<
		x
	)
	{
	}
```

mharsch, 2017-06-12:
The sleep functions have an overhead cost that is relatively fixed (but does vary by pi model).  e.g. calling usleep(0) takes about 72 microseconds on Rpi3, but takes about 130 microseconds on a Pi Model B.  Calling usleep(1) takes about 73 microseconds on Rpi3 and 131 microseconds on PI Model B (and so on).

ibmua, 2017-06-12:
You might want to include that in the docs 😉

closed 2017-10-04 by jperkin

mharsch, 2017-10-04:
Correction: the 130 microsecond value was for for raspi-1 (model B), not raspi-2.

Regarding the issue of timing in general: I've observed a significant difference in the execution time of rpio functions between the first call (slower) and subsequent calls (faster).  I suspect this variance is due to v8 recognizing the codepath and dynamically optimizing it for later calls.  I mention this in the context of usleep() because my attempts to pad the timing of communication with an i2c device have failed due to this variance.

I wonder if there is some way to coerce node into being more deterministic.

commit e27d1ad8 (2017-10-04): Document sleep call latency, fixes #18.