jperkin/node-rpio #19: Precision read of time duration of high pin

richpicking, 2016-09-05:
Hello,
This works great for a few projects but I'm struggling to get good precision timing of a pin going high with signal from an ultrasonic sensor..
(standard kit HC-SR04 - you send a high value to a trigger pin to tell it to measure the distance and it replies on an echo pin by setting a high value for the number of micro/nano seconds the ultrasound took to rebound off the nearest object).  You're supposed to measure the time of the pulse and calculate distance using the speed of sound.

I tried a read loop with simple conditions to catch the edges and get accurate time measurement from process.hrtime but I can't seem to get any level of consistency in results.  Have tried readbuf (like the example in the docs) and then try to guess the time between the edges (by measuring the total time to execute readbuf and dividing by buffer length - then multiplying by number of high values in the buffer)... but still can't seem to get reliably repeatable results.

I have some python code that runs on the same hardware which gives repeatable measurements but really wanted to stay in node.

Any suggestions of best way to get precision time measurement of a high pin pulse ?

Many thanks.

richpicking, 2016-09-15:
I wonder, is this something that I should be going to rpio2 for ?

jperkin, 2016-09-15:
Odd, I replied to this when you first raised the issue but it looks like it didn't show up.  Anyway, yes unfortunately `readbuf()` is the only way to get high resolution reads, and it can be a bit hit and miss when trying to align all the timings afterwards (as shown in my dht11 example) once you get back to JavaScript land.  I wonder if in your case the gap between executing `write()` and `readbuf()` is too long that you actually miss the response completely.

I don't know what rpio2 is, but if it's JavaScript then it'll suffer from exactly the same problems.

richpicking, 2016-09-15:
ah, thanks for the answer.
I do manage to see the high pin condition show up in the readbuf() output - I can count the number of 1s in the result set.. the problem is that I guess readbuf isn't working in a linear way - so for the same distance to measure, I see a different number of 1s every time - if you repeat the distance measure 10 times in a row, you get a few value counts that are close to each other but others it's way off - with no change in the physical position of the sensor... and they're too far out to be able to eg. read 20 times and take an average.

rpio2 seems to be an object oriented wrapper around rpio that can set events for edges of signals..  I'm not sure how they would be able to wrap with greater precision but will give it a go.
https://www.npmjs.com/package/rpio2

Many thanks again for your help.

astriffe, 2017-10-08:
@richpicking Have you found a solution to your problem? I'm at the same point where I'd like to get reliable results from my HC-SR04 using rpio.