jperkin/node-rpio #20: Use a "mock" mode for non-RPi platforms

DominikPalo, 2016-09-11:
It would be nice to add some mock mode for developing apps on other platforms than Raspberry Pi. 

E.g. I'm developing and debugging node app on PC (which uses also rpio module for some minor functionality), although in production it will be running on the Raspberry Pi. But, when I want to test the app on PC/Mac (without rpio functionality, of course) I have to comment out all usages of rpio, otherwise app crashes with an error:

> fs.js:640
>   return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
>                  ^
> 
> Error: ENOENT: no such file or directory, open '/proc/cpuinfo'

So my idea is that rpio can detect platform on which is running and on platforms other than Raspberry Pi, it will show some warning message like "Your platform is not compatible with module, RPIO functionality will not be available." instead of causing the node app to crash.

jperkin, 2016-09-28:
Yeh, interesting, I'll see what I can do.  This would also be useful for me if I want to test things in my usual OSX development environment.

tfiechowski, 2016-10-08:
I've also came across such problem. My needs have been so far very simple : just reading and polling 1/0 values on the pins. 
I've created a very simple rpio 'emulator' - you have file state.json where keys are the pin numbers and values represent values on that pins, basically just ones and zeroes. Emulator reads the file every 100ms and if a change occured it fires some callbacks. The interface for read, poll is the same as in regular rpio.

I use it with running node with `--elevator` option, and then in the code where I need the rpio I load rpio like this:

```
var EMULATOR_MODE = process.argv.some(function(arg) {
    if(arg == '--emulator') {
        return true;
    }
});

var rpio = null;
if(EMULATOR_MODE) {
    debug("Running with rpio emulator");
    rpio = require('../rpio_emulator/rpio_emulator.js')
} else {
    debug("Running with native rpio")
    rpio = require('rpio');
}
```

As I said, this a very simple solution for my needs, but feel free to check out and maybe use the code.

Zip with my code:
[rpio_emulator.zip](https://github.com/jperkin/node-rpio/files/517313/rpio_emulator.zip)

a2ron44, 2016-11-19:
I like this emulator, but there seems to be a problem with the emulator code.  Probably worthy of it's own code repository.

Getting an error saying this is not a function.  Not sure if a period is missing in this line or something else is off.  

 pollCallbacks.get("_" + pinChanged)(pinChanged, read(pinChanged));

adamreisnz, 2017-08-13:
Any progress on this issue?
It seems this error is triggered just by requiring the rpio library.

adamreisnz, 2017-08-13:
All it would take is not calling `setup_board()` immediately but giving this control to the developers, or wrapping the code `fs.readFileSync('/proc/cpuinfo')` in a try block and aborting the process.

jperkin, 2017-08-15:
Yes I'm working on a fix for this, just need to make sure it's done right. I'll try and get a fix out this week.

jperkin, 2017-08-19:
I've pushed support for mock mode, but haven't published it to NPM yet. Please can you folks give it a test and make sure it works ok?

If you're on unsupported hardware it'll be enabled automatically, defaulting to `raspi-3`. On all hardware you can pass a new `mock` option which should be set to one of the following:

    raspi-b-r1
    raspi-a
    raspi-b
    raspi-a+
    raspi-b+
    raspi-2
    raspi-3
    raspi-zero
    raspi-zero-w

depending on what hardware you want to emulate.

It seems to work ok for me both on supported and unsupported hardware, but I want to make sure it doesn't break anything before publishing a new release. Thanks.

adamreisnz, 2017-08-19:
Awesome, thanks for the swift action. I will take it for a spin.

adamreisnz, 2017-08-19:
Seems to work OK on non-raspberry pi hardware. However, I dislike that when you don't specify a mock explicitly logs console warnings. I would prefer an emitted warning event, so we can choose how to handle the warning ourselves. Will also take it for a spin on the R3.

jperkin, 2017-08-20:
Ok. The second part of this plan is to introduce some logging events which would show what would happen in mock mode (and also so you can see what is happening on real hardware), I'll see about handling it properly in that.

adamreisnz, 2017-09-15:
Could you please release the current version on npm with the mock mode?

jperkin, 2017-09-15:
I'm not entirely happy yet with how it's handled. Ideally what I want is to emit a warning event and have a default handler for it which will print out the warning if we are automatically running in mock mode, but allow the user to override it. I haven't yet figured out a clean way to do this, and until it's done right I don't want to commit to it.

The last couple of weeks have been busy with other stuff, I'll try to get to this soon.

adamreisnz, 2017-09-15:
Mind if I have a look at it?

jperkin, 2017-09-30:
Ok, turns out this was quite simple, and I've pushed a simple change in 0e5886a which should do what you need - it allows you to add e.g.

```
rpio.on('warn', function(){});
```

to override the default warn handler and do nothing, otherwise the default handler will print the warning message as before.  Let me know if this works ok for you and I'll push a new version to NPM.

adamreisnz, 2017-10-01:
Yep that should be fine, thanks!

adamreisnz, 2017-10-03:
It doesn't seem like it's working properly yet. I've tried the following:

```js
    //Bind warning handler
    rpio.on('warn', message => {
      this.onWarn(message);
    });

    //Initialize RPIO
    rpio.init({mock});
```

This happens in an initialization function after the constructor for the connection class is instantiated. However, I still see logs showing up in the console:

```
WARNING: Unable to auto-detect hardware
WARNING: Running in default mock mode (raspi-3)
```

The warnings are generated when the `init` method is called on rpio, so the listener should be bound at that stage already.

adamreisnz, 2017-10-03:
Upon checking the commit, it looks like my package manager might have used a cached version, I will try to force it to get the latest code from the repo.

adamreisnz, 2017-10-03:
Working as expected now! Thanks!

jperkin, 2017-10-04:
Ok I published 0.9.18 for mock support, I'll close this ticket now.

closed 2017-10-04 by jperkin