jperkin/node-rpio #21: HOWTO: Software SPI setting

Damaotimlin, 2016-09-20:
Hi All,

I wants to move my ruby/sinatra project to Node.js, but I'm facing an issue. Because of my hardware design is fixed and SPI wired(read MCP3008) I currently using was config by software SPI. I found here has SPI interface but all the example is for hardware SPI. I'm wondering how to use rpio to write a software SPI interface, thanks!

jperkin, 2016-09-20:
Hi!  There's nothing in the API dedicated to software SPI, however it should just be a case of using `.read()` and `.write()` across multiple pins and handling the clock yourself.

At some point I'll probably put some simple shift register examples into the repository, I have some code based on the older API [here](https://www.perkin.org.uk/posts/a-nodejs-powered-8-bit-cpu-part-four.html) which might be useful and just needs updating to use the current API.

Damaotimlin, 2016-09-20:
Hi @jperkin,

I'll take a look of the API you mentioned and see if I can figure it out by myself, due to I'm totally new to JS and not a program background guy, it might took me for a longer time. If I tested it out successfully, I'll share my code here. Before that, I'm looking forward to your examples, thanks !

jperkin, 2018-11-29:
Closing this one out for now.  I haven't had time to work on this for a while, but still hope to add some shift register examples at some point.

closed 2018-11-29 by jperkin