jperkin/node-rpio #22: Feature Request: Reverse Clock Polarity on PWM

sbarwe, 2016-09-20:
Sometimes it is neccessary to create an orthogonal PWM signals which is not easy to solve at high frequencies with inverters.

The reverse clock polarity feature of the BCM would be great to use here. May you extend the PWM at this e.g. that a call to rpio.open(12,PWM, 1) with init = 1 means that the clock is reversed?

example:

```
void bcm2835_pwm_set_mode(uint8_t channel, uint8_t markspace, uint8_t enabled, uint8_t reversepolarity)
{

...

  if (channel == 0)
    {
     ...
	 if (reversepolarity) 
		 control |= BCM2835_PWM0_REVPOLAR;
	 else
		control &= ~BCM2835_PWM0_REVPOLAR;
    }
  else if (channel == 1)
    {
 ...
	 if (reversepolarity) 
		 control |= BCM2835_PWM1_REVPOLAR;
	 else
		control &= ~BCM2835_PWM1_REVPOLAR;
    }

```

makuolgo, 2016-11-21:
That's what i thought as well but i have make installed:
make --version
```
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
Licence GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jperkin, 2018-11-29:
Sorry I haven't had the chance to look at this for a long time.  I'll leave this open for now, maybe someone else can submit a PR for this?