jperkin/node-rpio #25: Fails to install with npm 3.10.8 on node 6.9.1

makuolgo, 2016-11-21:
Hi!

Getting the following when running 'npm install rpio --save' on my Raspberry PI 3 (OS: Miniban).
Also getting the same error on Ubuntu 16.04 using the same version of Node.js.

```
> rpio@0.9.12 install /home/gobban/rpi-nodejs/node_modules/rpio
> node-gyp rebuild

gyp ERR! build error 
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:14:12)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:69:19)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:81:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:90:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/index.js:44:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/node_modules/isexe/access.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:123:15)
gyp ERR! System Linux 4.4.32-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/gobban/rpi-nodejs/node_modules/rpio
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN rpi-nodejs@1.0.0 No repository field.
npm ERR! Linux 4.4.32-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpio" "--save"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! rpio@0.9.12 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rpio@0.9.12 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the rpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs rpio
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls rpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/gobban/rpi-nodejs/npm-debug.log
```

npm-debug.log:
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'install', 'rpio', '--save' ]
2 info using npm@3.10.8
3 info using node@v6.9.1
4 silly loadCurrentTree Starting
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 silly fetchPackageMetaData rpio
8 silly fetchNamedPackageData rpio
9 silly mapToRegistry name rpio
10 silly mapToRegistry using default registry
11 silly mapToRegistry registry https://registry.npmjs.org/
12 silly mapToRegistry data Result {
12 silly mapToRegistry   raw: 'rpio',
12 silly mapToRegistry   scope: null,
12 silly mapToRegistry   escapedName: 'rpio',
12 silly mapToRegistry   name: 'rpio',
12 silly mapToRegistry   rawSpec: '',
[... 668 lines cut ...]
369 verbose stack     at ChildProcess.emit (events.js:191:7)
369 verbose stack     at maybeClose (internal/child_process.js:877:16)
369 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
370 verbose pkgid rpio@0.9.12
371 verbose cwd /home/gobban/dev/rpi-nodejs
372 error Linux 4.4.0-47-generic
373 error argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpio" "--save"
374 error node v6.9.1
375 error npm  v3.10.8
376 error code ELIFECYCLE
377 error rpio@0.9.12 install: `node-gyp rebuild`
377 error Exit status 1
378 error Failed at the rpio@0.9.12 install script 'node-gyp rebuild'.
378 error Make sure you have the latest version of node.js and npm installed.
378 error If you do, this is most likely a problem with the rpio package,
378 error not with npm itself.
378 error Tell the author that this fails on your system:
378 error     node-gyp rebuild
378 error You can get information on how to open an issue for this project with:
378 error     npm bugs rpio
378 error Or if that isn't available, you can get their info via:
378 error     npm owner ls rpio
378 error There is likely additional logging output above.
379 verbose exit [ 1, true ]
```

jperkin, 2016-11-21:
The relevant line is this one:

```
gyp ERR! stack Error: not found: make
```

You need to install make, usually via the build-essential package so that you get a compiler too.  Without this you won't be able to install any native modules.

closed 2016-11-21 by jperkin

makuolgo, 2016-11-21:
Thank you very much - worked fine after installing build-essential package.

Might be good to mention this in your README file.