jperkin/node-rpio #26: Request: Please add option for rpio.poll e.g. debounce, polling Interval

tosbodes, 2016-11-25:
Hello,
This module work very nice, and it's cover almost all use case, easy to install. But sometime rpio.poll call rapidly 

```
var rpio = require('rpio');

rpio.open(11, rpio.INPUT, rpio.PULL_UP);
rpio.open(7, rpio.OUTPUT, rpio.HIGH);

var output1State = rpio.HIGH;
setInterval(function(){
	console.log("------------")
	output1State = output1State == rpio.HIGH ? rpio.LOW : rpio.HIGH;
	rpio.write(7, output1State);
}, 1000);

function pollcb(pin)
{
    var state = rpio.read(pin) ? 'pressed' : 'released';
    console.log('Button event on P%d (button currently %s)', pin, state);
}
 
rpio.poll(11, pollcb);
```

OUTPUT
```
------------
Button event on P11 (button currently pressed)
------------
Button event on P11 (button currently released)
Button event on P11 (button currently released)
Button event on P11 (button currently released)
------------
Button event on P11 (button currently pressed)
------------
Button event on P11 (button currently released)
Button event on P11 (button currently pressed)
Button event on P11 (button currently released)
Button event on P11 (button currently released)
```

cullylarson, 2017-10-10:
This has been working pretty well for me:

```
function pushed(pin) {
    // wait a bit to make sure we aren't bouncing
    rpio.msleep(20)

    // if the pin is no longer down, assume it was just a bounce
    if(rpio.read(pin)) return

    // at this point, the pin is still down so assume it's a real push
    console.log('button pushed')
}

rpio.open(buttonPin, rpio.INPUT, rpio.PULL_UP)
rpio.poll(buttonPin, pushed, rpio.POLL_DOWN)
```

jperkin, 2018-11-29:
I quite like this workaround and may add something similar as a note to the documentation.

jperkin, 2018-11-29:
I added this in 04378da6dc3c75da1dd8d0b42886ef760eefd5aa

closed 2018-11-29 by jperkin