jperkin/node-rpio #29: aarch64 support?

cwilling, 2017-01-19:
After using rpio successfully on normal systems for while (thanks for providing it), I tried it on an aarch64 rpi3 board. It failed at setup_board() in lib/rpio.js because /proc/cpuinfo no longer contains a board Revision entry. I added code to look also in /proc/device-tree/model (where it finds "Rev 1.2" on my board) and I use that to set gpiomap to "v2plus" (which is what a plain 32bit system running on the same board sets it to).

That change seems to work OK, however now my app is unable to initialize the bcm2835 library (bcm2835_init: Unable to open /dev/gpiomem: No such file or directory). Evidently a newer version of the bcm2835 library is needed for aarch64. Do you know whether it exists and will you incorporate it any time soon?

Thanks,
chris

jperkin, 2017-01-19:
Interesting, thanks. Does your AArch64 kernel have gpiomem support built in? I'd find it odd if they moved the device node between kernel archs.

In any case, no there's no upstream bcm2835 changes in this area, but I will take a look on my RPi3 to try and reproduce. If you let me know which kernel you are using that'll help. Thanks.

cwilling, 2017-01-19:
Good catch! I didn't realize that gpiomem support was a kernel thing. The aarch64's /proc/config.gz shows CONFIG_BCM2835_DEVGPIOMEM (& related options) not set, whereas they are set in the plain 32bit version. The kernel builders have agreed to include them in the next update.

For interest, using kernel 4.9.3 from http://test.fatdog.eu/dev/alchemy/sarpi-aarch64/builds/4.9.3/.

Thanks,
chris

jperkin, 2017-01-19:
Alright, thanks. Could you paste the output of both `/proc/cpuinfo` and `/proc/device-tree/model`? I can at least fix that one relatively easily.

cwilling, 2017-01-19:
pi@aarch64test:~$ uname -a
Linux aarch64test 4.9.3-v8-arm64 #1 SMP PREEMPT Mon Jan 16 01:24:18 GMT 2017 aarch64 GNU/Linux
pi@aarch64test:~$ cat /proc/cpuinfo 
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

pi@aarch64test:~$ cat /proc/device-tree/model 
Raspberry Pi 3 Model B Rev 1.2

cwilling, 2017-01-20:
Just to confirm that a new kernel/modules with relevant config options set has fixed the lack of /dev/gpio* devices. Contents of /proc/cpuinfo & /proc/device-tree/model remain the same.

Thanks,
chris

jperkin, 2018-11-29:
Sorry I haven't got around to looking at this.  Is this still an issue with the latest kernels?

jperkin, 2020-01-13:
I think this should be fixed with cc33c3fff98a0d4d3346788cc5bcb98bc7a4aca2 and available in v1.2.3 and newer.

closed 2020-01-13 by jperkin