jperkin/node-rpio #31: Problem with install...

pi-plates, 2017-02-11:
When I run _sudo npm install rpio_ I get a slew of failures:

rpio.target.mk:83: recipe for target 'Release/obj.target/rpio/src/rpio.o' failed
make: *** [Release/obj.target/rpio/src/rpio.o] Error 1
make: Leaving directory '/home/pi/node_modules/rpio/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.38-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/rpio
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! rpio@0.9.15 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the rpio@0.9.15 install script.
npm ERR! This is most likely a problem with the rpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls rpio
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.38-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpio"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0

I installed gcc per the directions but there's no change to these error messages. I'm sure I'm doing something simple and stupid.

jperkin, 2017-02-11:
Is that the full output? Seems to be everything except for the actual error causing the problem.

pi-plates, 2017-02-12:
Ran it again and got the same:

pi@AGS:~ $ sudo npm install rpio
\
> rpio@0.9.15 install /home/pi/node_modules/rpio
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/rpio/build'
  CC(target) Release/obj.target/rpio/src/bcm2835.o
  CXX(target) Release/obj.target/rpio/src/rpio.o
In file included from ../src/rpio.cc:17:0:
../node_modules/nan/nan.h:324:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
rpio.target.mk:83: recipe for target 'Release/obj.target/rpio/src/rpio.o' failed
make: *** [Release/obj.target/rpio/src/rpio.o] Error 1
make: Leaving directory '/home/pi/node_modules/rpio/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.38-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/rpio
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! rpio@0.9.15 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the rpio@0.9.15 install script.
npm ERR! This is most likely a problem with the rpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls rpio
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.38-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "rpio"
npm ERR! cwd /home/pi
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log
npm ERR! not ok code 0

jperkin, 2017-02-12:
Ok that's better, it has the actual error message:

```
../node_modules/nan/nan.h:324:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
```

Unfortunately that means it's outside of my control, it's coming from the nan module which mine depends on.  I was under the impression that nan supported all known releases (it's the whole point of using it), but perhaps this is one where there are issues.  My module is automatically tested for each release, and the 0.10 build is currently passing <https://travis-ci.org/jperkin/node-rpio/jobs/197074050>

Probably the easiest way to get around this is to install a newer nodejs from upstream rather than using the older Debian package.  It might be that their version has issues that the upstream one doesn't (it's at least a lot older than the 0.10 version used by Travis).

Ediolot, 2017-03-18:
Try this as root [https://github.com/audstanley/NodeJs-Raspberry-Pi](https://github.com/audstanley/NodeJs-Raspberry-Pi).
It worked for me on a Raspberry Pi A+.

jperkin, 2017-10-04:
Closing this one as there's nothing I can really do about it.

closed 2017-10-04 by jperkin