jperkin/node-rpio #33: open with pud option may not be working

rgillan, 2017-03-03:
Hi,
On a RPi3 (haven't tested earlier versions) we have a range of I/O, both static and polled.  One of our polled inputs was acting as if it was floating as we were detecting a simple proximity switch to ground.  Any noise (or simply touching the lead) was triggering the poll detect.

	rpio.open(7, rpio.INPUT, rpio.PULL_UP);

We found that declaring the input, followed by the pud assignment (no other code changes) has fixed the issue.

	rpio.open(7, rpio.INPUT);
	rpio.pud(7, rpio.PULL_UP);

Not sure if it's an issue but more in case someone else needs a hand.  Will dig further once we get a chance.

jperkin, 2018-11-29:
Did you get a chance to investigate this? There were changes to this code but I don't know if they were before or after your tests.

rgillan, 2018-11-29:
Sorry, haven't tested in a while (our code uses the separate open/pud methods in sequence.  If we get a chance to test again I'll update (it's been working well for a long time, so we haven't bothered to change anything)

closed 2019-10-17 by jperkin