jperkin/node-rpio #36: How to use this lib with LCD communicated via SPI?

ysyyork, 2017-03-30:
I have a LCD screen communicated with the pi using HDMI and SPI (SPI for touches). Now every time I started to use this lib, it will take control of the SPI ports and the screen will automatically stop working. Any idea I can I leave SPI open for the screen?

jperkin, 2017-03-30:
Do you have any more details, e.g. the script you are running? SPI stops working even if you are only doing basic GPIO?

ysyyork, 2017-03-30:
@jperkin This is the code I used right now. rpio object is passed in via rpioClient. It's just simply rpioClient = require('rpio'). I checked all these pin numbers and I'm 100% percent sure I didn't touch SPI pins which are 19,21,23,24,26. The pins I use here are 16, 18, 22, 31, 32, 40. Now every time I initialize this class, it will reset my curser on the screen (the cursor goes to the upper left of the screen). Then if I touch the screen, the curser won't move. If I terminate this node program, the control of the curser gets back to the screen then. 

```
class PeripheralsClient {
    constructor(pinMapping, rpioClient) {
        if (pinMapping === void 0 ||
            pinMapping.magnet1 === void 0 ||
            pinMapping.magnet2 === void 0 ||
            pinMapping.ledRed === void 0 ||
            pinMapping.ledGreen === void 0 ||
            pinMapping.ledBlue === void 0 ||
            pinMapping.sensor === void 0) {
            throw new Error('Missing or invalid pinMapping');
        }
        else {
            this.pinMapping = pinMapping;
            this.rpio = rpioClient;
            this.rpio.open(this.pinMapping.magnet1, this.rpio.OUTPUT, this.rpio.HIGH);
            this.rpio.open(this.pinMapping.magnet2, this.rpio.OUTPUT, this.rpio.HIGH);
            this.rpio.open(this.pinMapping.ledRed, this.rpio.OUTPUT, this.rpio.HIGH);
            this.rpio.open(this.pinMapping.ledGreen, this.rpio.OUTPUT, this.rpio.HIGH);
            this.rpio.open(this.pinMapping.ledBlue, this.rpio.OUTPUT, this.rpio.HIGH);
            this.rpio.open(this.pinMapping.sensor, this.rpio.INPUT, this.rpio.PULL_UP);
        }
    }

    up() {
        this.rpio.write(this.pinMapping.magnet1, this.rpio.HIGH);
        this.rpio.write(this.pinMapping.magnet2, this.rpio.LOW);
    }

    down() {
        this.rpio.write(this.pinMapping.magnet1, this.rpio.LOW);
        this.rpio.write(this.pinMapping.magnet2, this.rpio.HIGH);
    }

    led(color) {
        switch (color) {
            case 'red': 
                this.rpio.open(this.pinMapping.ledRed, this.rpio.OUTPUT, this.rpio.HIGH);
                this.rpio.open(this.pinMapping.ledGreen, this.rpio.OUTPUT, this.rpio.LOW);
                this.rpio.open(this.pinMapping.ledBlue, this.rpio.OUTPUT, this.rpio.LOW);
                console.log('red light');
                break;
            case 'green':
                this.rpio.open(this.pinMapping.ledRed, this.rpio.OUTPUT, this.rpio.LOW);
                this.rpio.open(this.pinMapping.ledGreen, this.rpio.OUTPUT, this.rpio.HIGH);
                this.rpio.open(this.pinMapping.ledBlue, this.rpio.OUTPUT, this.rpio.LOW);
                console.log('green light');
                break;
            case 'blue':
                this.rpio.open(this.pinMapping.ledRed, this.rpio.OUTPUT, this.rpio.LOW);
                this.rpio.open(this.pinMapping.ledGreen, this.rpio.OUTPUT, this.rpio.LOW);
                this.rpio.open(this.pinMapping.ledBlue, this.rpio.OUTPUT, this.rpio.HIGH);
                console.log('blue light');
                break;
            case 'white':
                this.rpio.open(this.pinMapping.ledRed, this.rpio.OUTPUT, this.rpio.HIGH);
                this.rpio.open(this.pinMapping.ledGreen, this.rpio.OUTPUT, this.rpio.HIGH);
                this.rpio.open(this.pinMapping.ledBlue, this.rpio.OUTPUT, this.rpio.HIGH);
                console.log('white light');
                break;
            default:
                console.log('color not supported');
                break;
        }
    }

    isSensorBlocked() {
        return this.rpio.read(this.pinMapping.sensor) === 0;
    }

    closeAll() {
        this.rpio.close(this.pinMapping.magnet1);
        this.rpio.close(this.pinMapping.magnet2);
        this.rpio.close(this.pinMapping.ledRed);
        this.rpio.close(this.pinMapping.ledGreen);
        this.rpio.close(this.pinMapping.ledBlue);
        this.rpio.close(this.pinMapping.sensor);
    }
}
```

ysyyork, 2017-03-30:
@jperkin Sorry, my bad. I find the issue. I thought the screen only uses SPI but after a careful look, I find it also uses two GPIO. One is conflicted with the other app.

closed 2017-03-30 by ysyyork

jperkin, 2017-03-30:
Ok, no problem! Thanks for using the library ;)

ysyyork, 2017-03-31:
really solid library 👍