jperkin/node-rpio #39: Is it possible to change polling window?

nueverest, 2017-05-07:
In the readme it says,

> Interrupts aren't supported by the underlying hardware, so events may be missed during the 1ms poll window.

Does this mean that the pin is poll at a frequency of 1kHz i.e. once per 1ms?

Is it possible for me to reduce or increase the sampling frequency?

jperkin, 2017-05-08:
JavaScript does not (as far as I'm aware) support sub-millisecond timing, so unfortunately no way to increase the sampling frequency.  There's no support at the moment for reducing the frequency, as I figured people could do that in their app if necessary and couldn't think of a reason why you'd want to, but I can add an option for this if you can show there's a need (or you can just change the hardcoded millisecond option at https://github.com/jperkin/node-rpio/blob/master/lib/rpio.js#L491)

nueverest, 2017-05-08:
A reason to reduce the sampling frequency is to increase efficiency.  Say I am polling 10 GPIO pins each at 1kHz.  How do I determine the CPU impact?

jperkin, 2017-05-08:
It would be worth benchmarking, but I don't imagine it would make a huge amount of difference how many pins you are polling - the function is only called once either way.  I'll leave the ticket open though and try to get some numbers on it at some point.  I think I still have a working original Raspberry Pi around somewhere to exacerbate the CPU issue.

NateZimmer, 2019-06-16:
Forgive my ignorance of linux drivers. So I know `rpi-gpio.js` uses `epoll` to detect changes in GPIO. Would this be any faster/more efficient than what you are doing?

adminy, 2022-02-26:
Hi,
I'm not sure whether my use case extends this issue or it should be part of new issue, but

Currently there are no RF libraries that work with pure node in js. 
for RF the frequency is 433 MHZ I think, so its a lot faster than this.
I was hoping I could use this library, but Honestly I was expecting I'd even have cycles to spare with this library. Maybe 🤔 I'm just dumb and I don't see it but how exactly does 433 mhz actually work, maybe its send speed is different on the actual chip than the gpio, in that case I could totally use rpio for sending and receiving 433 mhz using the devices I have:
MX-RM-5V and MX-FS.

It says they have a 4KB/s transfer rate. 4 * 1,024 * 8 = 32,768 bits to send or receive in a second right?

So then ... I could use this library for that purpose? I'll try, let me know if I'm not making too many assumptions here or got something wrong.
Thanks 🙏🏻 

(ps. future work could be to make an API call for this?! since that would be a nice addition to have RF and speed boost maybe 🤔)