jperkin/node-rpio #40: Led pulse not working

nevaldas, 2017-05-26:
Hey,

Running the given example (without any modification) on Raspberry Pi 3 Model B lights the LED and does not pulse it. Using Piscope I see that value of 1 is set for the whole duration - from open to close, no matter what I pass to pwmSetData. I even tried to change Clock Divider values and set different range with no success. Any ideas what could be wrong? I use GPIO18 (pin 12), as used in the example.

Another thing - it would be really nice if one could set duty cycle and forget about setting those other values which I have no idea about yet.
I have managed to pulse the led only with pigpio's (node.js) and RPi.GPIO's (python) duty cycle methods.

igna92ts, 2017-09-12:
Currently having the same issue

mamacker, 2017-11-12:
I'm having the same issue on a Raspberry PI 3 Model B, as well.  Stays high, then shuts off when the pin is changed to an input.

I'll keep digging.