jperkin/node-rpio #42: warn against mixing rpio SPI with spi-bcm2835 + spidev

mharsch, 2017-06-19:
rpio SPI can't coexist with the regular linux SPI kernel module  (at least for a given spi bus).  Running a rpio SPI script on a system with spi-bcm2835 loaded will corrupt the state of the kernel module and break any subsequent calls to it's interface (silently).  We should include a warning in the documentation.

jperkin, 2017-06-19:
Yeh this is the same as regular GPIO if you try to mess via `/sys/class` at the same time. I'll add a note.

mightydread, 2020-01-23:
Hi... can you explain a bit more? I'm trying to write an script with this library and have issues with the spi on a Rpi 3 with the  spi-bcm2835. but if I blacklist it no spi devices appear under /dev