jperkin/node-rpio #43: RPI 1

bahamut657, 2017-07-05:
I try your code in RaspberryPI 1.
Using mapping = 'gpio' seems does not work.
Reverting to mapping = 'physical' work correctly.

jperkin, 2017-07-05:
Any more details? In what way does it not work? Have you double checked your pin configurations? Are there any specific pins which are not working? Thanks.

bahamut657, 2017-07-05:
This is the code:
const RPIO = require("rpio");
RPIO.init({
        mapping:"gpio" 
});  /** IF I REMOVE THIS INIT AND CHANGE THE PIN NUMBER TO 12 IT WORKS **/

RPIO.open(1, RPIO.OUTPUT, RPIO.LOW);
for (var i = 0; i < 5000; i++) {
        /* On for 1 second */
        RPIO.write(1, RPIO.HIGH);
        RPIO.sleep(1);

        /* Off for half a second (500ms) */
        RPIO.write(1, RPIO.LOW);
        RPIO.msleep(500);
}

jperkin, 2017-07-05:
GPIO1 is pin5 not pin12, check the mapping you have for your hardware revision. I find <http://raspi.tv/wp-content/uploads/2014/07/Raspberry-Pi-GPIO-pinouts.png> to be quite a useful one.

closed 2017-07-06 by bahamut657