jperkin/node-rpio #44: How can i implement below code in this library

w0w, 2017-07-10:
I want to implement below python code using this library, if it is possible then how to go about it.

```
import smbus
import time

def getTempC():
    bus = smbus.SMBus(1)
    bus.write_byte(0x40, 0xF3)
    time.sleep(0.3)
    data0 = bus.read_byte(0x40)
    data1 = bus.read_byte(0x40)
    cTemp = ((data0 * 256 + data1) * 175.72 / 65536.0) - 46.85
    print ("Temperature in Celsius is : %.2f C" %cTemp)
    return cTemp
```

jperkin, 2017-07-17:
Could you let me know what you've tried and what doesn't work so far? Is there any part of the documentation that isn't clear? I'd like to ensure that the documentation experience is good for new users, and if the current documentation doesn't explain how to achieve this currently then I'd like to fix that.  Thanks.

w0w, 2017-07-17:
@jperkin 

below is the code i have tried and both data1 and data2 have value 0, console.log prints -46.85

```

var rpio = require('rpio');

rpio.i2cBegin();
rpio.i2cSetSlaveAddress(0x40);

var txbuf = new Buffer([0x40, 0xF3]);
var rxbuf = new Buffer(64);

rpio.i2cWrite(txbuf);           

/* Reads 2 bytes */
data0 = rpio.i2cRead(rxbuf, 2)
data1 = rpio.i2cRead(rxbuf, 2)

console.log(((data0 * 256 + data1) * 175.72 / 65536.0) - 46.85);

rpio.i2cEnd();

```

I might be totally off from what needs to be done since this is the first time i am working on rpi with javascript.

jperkin, 2017-07-18:
Few things:

* Your `txbuf` looks incorrect, in the python code you are sending `0xF3` but in the node code you are sending `0x40 0xF3`.
* You probably want to add an `rpio.msleep(300)` after the write to match the python code and ensure the device has accepted the command.
* The data read is in `rxbuf`, not in `data0` and `data1` (which you have assigned as the return codes of those functions).  You are also overwriting `rxbuf` in the second call. You likely just need one call and then use `rxbuf[0]` and `rxbuf[1]` as the values.

Let me know if anything was unclear from the documentation and I'll try to improve it.  Thanks.

closed 2018-11-29 by jperkin