jperkin/node-rpio #46: Supporting Allwinner chips

lubino, 2017-08-17:
I understand that this is more like a feature request, because this library is "Raspberry Pi GPIO interface", but I really like this NPM module and I am using quite long time and I am not using only RPI. So I extended this project and implemented the C++ part for Allwinner's H3 and H5 chips (nanopi and orange pi boards). The module finds out what board you are using and uses right code using the same, unchanged node interface. I would like to publish it to NPM, so I am asking all of you guys, do you think that I should publish it under different NPM module name, or just ask for pull request to this original "rpio" NPM module? Thanks a lot.

jperkin, 2017-08-17:
It sounds like this would be a great addition to the library. Could you post your code somewhere so I can take a look? If as you say they are compatible boards then I see no reason why it couldn't be integrated.

lubino, 2017-08-17:
The GPIO pinouts are compatible, but they aren't using the same bcm2835 chip. Allwinner is a bit different. Ok, I would fork your repo and do the changes. I really like Orange Pi because of SATA port and Nanopi because of small dimensions and 1gbps LAN port. But to be hones, I don't like my implementation so much, yours is so much cleaner. I've used WiringPi C++ code to get the GPIO working on this boards.

lubino, 2017-09-01:
Sorry that I didn't respond for 2 weeks. I joined the WiringPN and WiringOP code into original WiringPi. It is more clear now and I also tested all the boards I have (Nano Pi with H3 and H5 chips as well as Orange Pi with H3 chip). It's working so I am going to create a pull request for you.

vanarebane, 2020-08-08:
What about merging the pull request?

jperkin, 2020-08-08:
I have support for this locally, I just need to finish it off when I have a bit of spare time.

vanarebane, 2020-08-08:
I'm not very great with C, have only skills to modify existing code. I do great in JS. Anything I can help with?

lubino, 2020-08-08:
Hi, maybe I can help. This is what I created, I think it is going to celebrate 3rd year, happy birthday soon 🙂

So, please, just tell me what to do and I would do that.

vanarebane, 2020-08-08:
@lubino I saw your fork of rpio with NanoPi and Orange Pi support. Any chance it can work with Banana Pi? I have one BPi-M2 Zero with Allwinner H2+ chip, and I'm desperate to get the GPIO working with Node.JS.

lubino, 2020-08-08:
I’ve never tried that, but it should be.

jperkin, 2020-08-08:
I wrote my own code which worked across a bunch of different sunxi-based boards that I bought.  I'll try to at least get a diff out over the next few days so people can test before I integrate.

vanarebane, 2020-08-08:
I traced the pinmap for Banana Pi M2 Zero

PINMAP_BPI_M2_ZERO: [
    -1,
    -1, -1,
    12, -1,
    11, -1,
     6, 13,
    -1, 14,
     1, 16,
     0, -1,
     3, 15,
    -1, 68,
    64, -1,
    65,  2,
    66, 67,
    -1, 71,
    19, 18,
     7, -1,
     8, -1,
     9, -1,
    10, -1,
    17, 21,
    -1, 20
],

@jperkin I'd be very grateful if you share, I can test my BPi if you get it diff out

vanarebane, 2020-08-21:
Hi, is it possible for you to share something? I can try fix bugs myself if you can't spare time. I have a expensive project on hold and I can't switch boards now. Thanks!

jperkin, 2020-08-21:
I'll see if I can dig my boards out over the next few days and put something in a branch.

vanarebane, 2020-08-21:
Hi, can I contribute? I'm writing BananaPi M2 Zero pinout to rpio.js at the moment and doing research on how to differentiate different H3 / H2+ boards (like Orange, Banana and Nano)

vanarebane, 2020-08-23:
Damn, well, it's yet another obstacle. Either I set up software SPI or help with hardware SPI. Any idea if the Broadcom and Sunxi SPI protocols are similar? I mean how much Boradcom script can be reused for Sunxi? Let me know if you intend to work on the I2C/PWM/SPI support and if I can help in anything. I have time to track down documentation and I can write C.

jperkin, 2020-08-23:
From a brief look it seemed similar, just need to figure out the correct memory ranges and wire everything up.  As you can tell by the delay in this ticket I'm not great at finding time to work on this these days, but yeh I'd still love to have support for it and was planning to add it.

jperkin, 2020-08-23:
As for documentation, there's the existing pull request based on WiringPi which I think already support this, I just didn't like the WiringPi code and wanted to write my own version that was a lot more understandable (to me at least).  Most of the code out in various projects for these chipsets seem to all be randomly copy/pasting bits from each other, so I mostly just went directly to the Allwinner SoC datasheet pdf that I managed to find (`Allwinner_V40_Datasheet_V1.0.pdf`) and implemented my own clean-room code.

vanarebane, 2020-08-24:
You mean [this library](https://github.com/BPI-SINOVOIP/BPI-WiringPi2/tree/master/wiringPi)?
I'll give this existing WiringPi pull request a try and if it works, it's a good place to look for working example.