jperkin/node-rpio #48: "Unable to determine board revision"

felixgalindo, 2017-08-18:
Getting "Unable to determine board revision" for CM3 raspberry pi.

felixgalindo, 2017-08-18:
Hardware	: BCM2835
Revision	: a020a0

felixgalindo, 2017-08-18:
Added support for Compute Module 3 revision number.

jperkin, 2017-08-18:
Thanks, I'll take a look at this. Is it expected that most people will use the expansion header module? Adding support should be straight-forward but I'll need to implement a physical pin mapping, so if it's expected that everyone will have the J5 and J6 60-pin headers then I can write a single mapping.

jperkin, 2018-11-29:
Have committed this as-is for now.  I'd still like to have proper Compute Module support, but this should at least get things working for you out of the box for now.  Thanks!

closed 2018-11-29 by jperkin