jperkin/node-rpio #53: RPIO in browser (run by javascript): Uncaught TypeError: exists is not a function

robson1014, 2017-09-06:
Hello,
I have to have RPIO in browser so as to control GPIO by javascript but not node.js. To do that, i used browserify. Everything compuled but after running test html with attached javascript bundle code i got error:  `bundle.js:880 Uncaught TypeError: exists is not a function
    at Function.getRoot (bundle.js:880)
    at bindings (bundle.js:782)
    at Object.<anonymous> (bundle.js:42)
    at Object.2._process (bundle.js:720)
    at s (bundle.js:1)
    at bundle.js:1
    at Object.1.rpio (bundle.js:2)
    at s (bundle.js:1)
    at e (bundle.js:1)
    at bundle.js:1
getRoot @ bundle.js:880
bindings @ bundle.js:782
(anonymous) @ bundle.js:42
2._process @ bundle.js:720
s @ bundle.js:1
(anonymous) @ bundle.js:1
1.rpio @ bundle.js:2
s @ bundle.js:1
e @ bundle.js:1
(anonymous) @ bundle.js:1`

I tried to change `exists` to `fs.exists` or `fs.existsSync` but the same error occupied.

Please help.
Thanks in advance.

jperkin, 2017-09-07:
Thanks for the issue. I'm not familiar with browserify, but I can't find any documentation which explains how it is supposed to work with native modules. This module requires a binary addon for `mmap()` access to the Broadcom device memory region, and I'm not sure how that's going to work in a browser context. Are you sure this is supported?

robson1014, 2017-09-07:
I am not sure, but I love your library, and i'd like to use it in web browser (javascript). Can you tell me is it possible ? Thanks for your fast reply.

jperkin, 2017-09-08:
I honestly have no idea, sorry. It's probably worth asking the browserify community as they will be much more aware of what it supports and what it can't access from a browser context. In terms of using the library from a browser, it wouldn't be much work to write a small app which provides an interface over a websocket to control a node.js process from a browser. I assume this is what you are hoping to achieve?

I'll leave this issue open for now as it might be worth providing an example of how to do this.

dlemper, 2017-10-12:
You cannot do this, as all the browsers do sandboxing. You cannot even get a file with javascript in your browsers console by simply calling a function as the user has to explicitly select a file with the file browser of your operating system. And if you can't even get the contents of a file, you will definetely not get access to /dev/mem or /dev/gpiomem (these are used internally by rpio).
Sorry for that.

What you can do is, write a simple 'server side' application using expressjs for example and build a really simple HTTP API with it around rpio. then you can use this API in your browser to access the GPIOs that way. If you would like to get example code, I'll try to catch some time making one.

jaqua, 2018-10-08:
As @dlemper already explained, you have to use an api as rpio has to be called on server side. You want to get access to the hardware rpi pins, so that is not possible to be done on the client side, though ES6 doesn't make sense.

jperkin, 2018-11-29:
Closing this one out, rpio is never going to work in a browser context.

closed 2018-11-29 by jperkin