jperkin/node-rpio #54: Thrown: GPIO12 is currently in use by /sys/class/gpio

japita-se, 2017-09-16:
I think this function is not necessary in rpio.js

`function check_sys_gpio(pin)`

In fact, in RPI3  + nodejs 8 we can't use rpio.
If I comment that function all is fine.

jperkin, 2017-09-16:
It's required so that the module does not conflict with any process that might be using the `/sys` interface. You need to unexport the pins via `/sys` first, so that the gpio state cannot be corrupted.

closed 2017-09-16 by jperkin

japita-se, 2017-09-20:
Uhm... can you tell me how to unexport?
Because, if I comment that function everything works.
g

jperkin, 2017-09-20:
You echo the pin number to the `/sys/class/gpio/unexport` control file. Pins are not exported by default, so something on your system has enabled them without disabling them again afterwards, hence the conflict.  You can read more about the sysfs interface here <https://www.kernel.org/doc/Documentation/gpio/sysfs.txt>

This module uses a different interface to control the pins, which is why we bail out if the sysfs interface is in use. While it may happen to work ok by commenting out the check, it's only due to pure luck.  You can't use both interfaces simultaneously and expect things to work correctly.

japita-se, 2017-09-20:
Ok, thank you very much for your help