jperkin/node-rpio #56: rpio.poll(...) callback executed after polling stopped.

newmanw, 2017-10-04:
```
function cb() {
  console.log('on poll');
}

rpio.poll(pin, cb);
...
rpio.poll(pin, null);
// The cb() function is often called after we deregister
```

Would it be possible to prevent callbacks on a pin from being sent if poll is deregistered?

DaniVarga, 2017-10-14:
Same problem here, got error:
` (node:10058) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 pin23 listeners added. Use emitter.setMaxListeners() to increase limit
`

DaniVarga, 2017-10-15:
Here is a little demonstration

```
var rpio = require('rpio')
rpio.init({ mapping: 'gpio' });

var pin = 23;

function foo () {
    console.log('foo');
}

rpio.open(pin, rpio.INPUT);
rpio.poll(pin, foo, rpio.POLL_HIGH);   
for (i = 0; i < 20; i++) {  
    rpio.poll(pin, null, rpio.POLL_HIGH);  
    rpio.poll(pin, foo, rpio.POLL_HIGH);     
}

rpio.open(pin, rpio.OUTPUT);
rpio.write(pin, rpio.LOW)
```

The script generates following output: 

> $ sudo node test.js
> (node:12646) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 pin23 listeners added. Use emitter.setMaxListeners() to increase limit
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo
> foo

I switched from 0.9.16 to 0.9.19, on 0.9.16 the problem was not present afaik.

DaVarga, 2018-06-03:
Still facing this problem. Any news about this?

Wouter0100, 2018-09-18:
@DaVarga could you test #77? It works on my end.

DaVarga, 2018-09-20:
@Wouter0100 I am very busy at the moment but I will test it tonight.

commit 9d6b5c34 (2018-10-04): Fix poll removals after warn event changes.

Resolves #56, patch submitted by Wouter van Os in #77.

Release 0.9.23.
closed 2018-10-04 by jperkin