jperkin/node-rpio #57: dht11 example produces NaN for all values logged

riatzukiza, 2017-11-01:
In addition to the NaNs, upon logging tmp and buff both, they appear to be all `1`'s, I get a similar result if I run the program with out anything connected.

riatzukiza, 2017-11-03:
I've tried this lib, an example off node-rpio, and the adafruit dht lib, and I am not able to get a reading off of this sensor, I think that it may just be broken.

riatzukiza, 2017-11-04:
I got another sensor and tried it, the example as it exist in the rep provides all NaNs, I changed a bit and I got:
```
Read failed:
    chk = 0, chksum = NaN\
    humidity: 27.6 temperature:80.NaN (0 = NaN)
```
Where the numbers fluctuate quite wildly, but the temperature dec value is usually NaN.
Here are the significant changes I made:
```js
rpio.open(pin, rpio.OUTPUT);
rpio.write(pin, rpio.HIGH);
rpio.msleep(500);
rpio.write(pin, rpio.LOW);
/*
 * The datasheet says 18us, but we need to account for the JavaScript function
 * call overhead.  Trial and error suggests this is a working value.
 */
rpio.msleep(20);

rpio.write(pin, rpio.HIGH);

rpio.usleep(80);
rpio.write(pin, rpio.LOW);
rpio.mode(pin, rpio.INPUT);

rpio.usleep(100);

/*
 * Switch to input mode and read as fast as possible into our buffer.
 */
rpio.readbuf(pin, buf);

```

atrovato, 2018-10-02:
Any news on this topic ?

atrovato, 2018-10-06:
Solution is here : 

`

    /*
     * Initiate the MCU sequence.
     */
    rpio.open(pin, rpio.OUTPUT, rpio.PULL_UP);
    rpio.sleep(1);
    rpio.write(pin, rpio.HIGH);
    rpio.msleep(10);
    rpio.write(pin, rpio.LOW);

    /*
     * The datasheet says 18us, but we need to account for the JavaScript function
     * call overhead.  Trial and error suggests this is a working value.
     */
    rpio.msleep(20);
    rpio.write(pin, rpio.HIGH);

    /*
     * Switch to input mode and read as fast as possible into our buffer.
     */
    rpio.mode(pin, rpio.INPUT);
    rpio.usleep(100);
    rpio.readbuf(pin, buf);
    rpio.close(pin);

`

jperkin, 2020-01-27:
I rewrote the dht11 example in 049e1878decb5f29ae26cbeb6e7ed3c00e3aee11 but forgot about this ticket, if there are still issues with the code let me know, it worked pretty well for me across a range of devices though.

closed 2020-01-27 by jperkin