jperkin/node-rpio #65: binding.js "the request of a dependency is an expression" with require('rpio')

Guttata, 2018-02-18:
Hi everyone,

I'm pretty new to nodejs. I'am building an app using create react app tool.
I installed rpio with "npm install -s rpio". The dependancy is written in _package.json_. The rpio folder is well created inside _<myprojectpath>/node_modules_.

I created a _DriveController.js_ with : 
```
export default function MoveForward() {
   var rpio = require('rpio');
   rpio.open(7, rpio.OUTPUT, rpio.LOW);

   for (var i = 0; i < 5; i++) {
         /* On for 1 second */
         rpio.write(7, rpio.HIGH);
         rpio.sleep(1);
  
         /* Off for half a second (500ms) */
         rpio.write(7, rpio.LOW);
         rpio.msleep(500);
   }

}

```
I import this file inside _DrivingMode.js_ with `"import MoveForward from './DriveController';"` and then in my code i call the MoveForward() function.

I got the following warnings at build with `npm start` command
```
Compiled with warnings.

./node_modules/bindings/bindings.js
81:22-40 Critical dependency: the request of a dependency is an expression

./node_modules/bindings/bindings.js
81:43-53 Critical dependency: the request of a dependency is an expression
```

and when the MoveForward() function is executed it crashes with : 

```
TypeError: exists is not a function
Function.getRoot
node_modules/bindings/bindings.js:158
  155 |   // Avoids an infinite loop in rare cases, like the REPL
  156 |   dir = process.cwd()
  157 | }
> 158 | if (exists(join(dir, 'package.json')) || exists(join(dir, 'node_modules'))) {
  159 |   // Found the 'package.json' file or 'node_modules' dir; we're done
  160 |   return dir
  161 | }
View compiled
bindings
node_modules/bindings/bindings.js:60
  57 | 
  58 | // Get the module root
  59 | if (!opts.module_root) {
> 60 |   opts.module_root = exports.getRoot(exports.getFileName())
  61 | }
  62 | 
  63 | // Ensure the given bindings name ends with .node
View compiled
(anonymous function)
node_modules/rpio/lib/rpio.js:17
  14 |  * OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.
  15 |  */
  16 | 
> 17 | var binding = require('bindings')('rpio');
  18 | var fs = require('fs');
  19 | var util = require('util');
  20 | var EventEmitter = require('events').EventEmitter;
View compiled
./node_modules/rpio/lib/rpio.js
http://192.168.1.15:3000/static/js/bundle.js:58160:30
__webpack_require__
/home/pi/dev/wavestone-autonomous-car/webpack/bootstrap 67d44566ab0d03daf48e:678
  675 | };
  676 | 
  677 | // Execute the module function
> 678 | modules[moduleId].call(module.exports, module, module.exports, hotCreateRequire(moduleId));
  679 | 
  680 | // Flag the module as loaded
  681 | module.l = true;
View compiled
fn
/home/pi/dev/wavestone-autonomous-car/webpack/bootstrap 67d44566ab0d03daf48e:88
  85 | 		console.warn("[HMR] unexpected require(" + request + ") from disposed module " + moduleId);
  86 | 		hotCurrentParents = [];
  87 | 	}
> 88 | 	return __webpack_require__(request);
  89 | };
  90 | var ObjectFactory = function ObjectFactory(name) {
  91 | 	return {
View compiled
MoveForward
src/DrivingMode/DriveController.js:2
  1 | export default function MoveForward() {
> 2 |    var rpio = require('rpio');
  3 |    rpio.open(7, rpio.OUTPUT, rpio.LOW);
  4 | 
  5 |    for (var i = 0; i < 5; i++) {
View compiled
Collection.<anonymous>
src/DrivingMode/JoystickDrivingMode.js:37
  34 |   this.setState({
  35 |     move: data,
  36 |   })
> 37 |   MoveForward()
  38 | })
  39 | .on('pressure', (evt, data) => {
  40 |   this.setState({
View compiled
▶ 9 stack frames were collapsed.

```
I don't understand the problem with the line `var binding = require('bindings')('rpio');` in _rpio.js_

If i run DriveController.js alone with `node DriverController.js` it's working well.

Thanks for your help.

jperkin, 2018-02-19:
This error:

```
TypeError: exists is not a function
```

is very similar to jperkin/node-rpio#53, i.e. trying to run node-rpio inside a browser context. Is this what you're trying to do? That's not going to work. node-rpio needs to run directly on the hardware.

Guttata, 2018-02-19:
Hi thanks for the quick reply. I think you are right.

I will implement a server / client WebSocket to workaround this.

closed 2018-02-19 by jperkin

dlemper, 2018-02-19:
Hi!
The implementation of your websocket application may be a nice example for other people like you, who want to get the gpio pins accessable from the web.. maybe @jperkin would include it, if you create a small pull request