jperkin/node-rpio #66: Implementing IR protocol

dlemper, 2018-03-05:
Hi,
I thought about implementing a very old IR protocol (ITT, https://www.sbprojects.net/knowledge/ir/itt.php), as I need it for my old receiver/amplifier.

Simply said, the protocol is based on 1's and 0's, where 1's are defined as a burst of 10us and then silence for 190us (200us cycle) and 0's are defined as 10us burst with a following silence of 90us.
The protocol specifies a preliminary pulse of 300 us, the a start pulse (0), then address bits (4 0's or 1's), then command (6 0's or 1's) and then stop pulse.

Now the tricky part: I'd like to do this on a first gen RPi, where the rpio.usleep() function takes 131us to execute..
I had three ideas to implement this thing:
1. PWM, somehow..
2. rpio.writebuf() and measure the time it takes for the writes (calibration), then build a buffer with more or less correct timings
3. add a function to rpio.cc, that is like writebuf but with loop that waits for given period

Any suggestions on this?

Thanks in advance!

dlemper, 2018-03-16:
OK this seems to bevnearly impossible.. I‘m now solving this problem with an arduino connected to the RPi.

closed 2018-03-16 by dlemper

jperkin, 2018-03-16:
Yeh sorry, JavaScript just isn't a great choice for sub-millisecond requirements, and while I could add extra functions to the C code for specific features I think ultimately it's best to keep things simple and just use native code where required.