jperkin/node-rpio #69: How to access object value inside the poll callback function ?

Guttata, 2018-03-23:
Hi,

Thanks again for this awesome library.

I'm facing a conception issue using the poll function that i want to use to count encoder ticks on my robot.

Inside countTickPinA function, "this" does not refer to my Encoder object.
How can i do to access its attributes (this.pinB, this.nbTicks) and methods (this.setTraveledDistance) ?
Is it possible to pass other params to the callback function of poll ?

Thanks a lot

```
//Encoder class and constructor
function Encoder(name, pinA, pinB) {
	this.pinA = pinA
	this.pinB = pinB;
	this.nbTicks = 0;
	this.traveledDistance = 0; //distance in meter

	rpio.open(pinA, rpio.INPUT, rpio.PULL_DOWN);
	rpio.poll(pinA, this.countTickPinA);
		
	rpio.open(pinB, rpio.INPUT, rpio.PULL_DOWN);
	rpio.poll(pinB, this.countTickPinB);
}

//callback function of the rpio.poll
Encoder.prototype.countTickPinA = function (pinA){
	console.log("PIN A : ", pinA );
	console.log("PIN B : ", this.pinB);
	
	//Returns either 0 for LOW or 1 for HIGH
	var statePinA = rpio.read(pinA);
	var statePinB = rpio.read(this.pinB);
	

	// look for a low-to-high on channel A
	if (statePinA == 1){
		// check channel B to see which way encoder is turning
		if (statePinB == 0) {
			this.nbTicks = this.nbTicks - 1; //CW
		}
		else {
			this.nbTicks = this.nbTicks + 1; // CCW
		}
	}
	// must be a high-to-low edge on channel A
	else {
		// check channel B to see which way encoder is turning
		if (statePinB == 1) {
			this.nbTicks = this.nbTicks - 1; //CW
		}
		else {
			this.nbTicks = this.nbTicks + 1; //CCW
		}
	}
	console.log(this.nbTicks);
	this.setTraveledDistance();
}
```

Guttata, 2018-03-23:
Hi,

I finally workaround like this : 
is it clean to do like this ?

```
function Encoder(name, pinA, pinB) {
	this.name = name;
	this.pinA = pinA
	this.pinB = pinB;
	this.nbTicks = 0;
	this.traveledDistance = 0; //distance in meter

	var encoder = this;

	rpio.open(pinA, rpio.INPUT, rpio.PULL_DOWN);
	rpio.poll(pinA, function countTickPinA(pinA){
		//Returns either 0 for LOW or 1 for HIGH
		var statePinA = rpio.read(pinA);
		var statePinB = rpio.read(encoder.pinB);
		
		// look for a low-to-high on channel A
		if (statePinA == 1){
			// check channel B to see which way encoder is turning
			if (statePinB == 0) {
				encoder.nbTicks = encoder.nbTicks - 1; //CW
			}
			else {
				encoder.nbTicks = encoder.nbTicks + 1; // CCW
			}
		}
		// must be a high-to-low edge on channel A
		else {
			// check channel B to see which way encoder is turning
			if (statePinB == 1) {
				encoder.nbTicks = encoder.nbTicks - 1; //CW
			}
			else {
				encoder.nbTicks = encoder.nbTicks + 1; //CCW
			}
		}
		console.log(name, " ", encoder.nbTicks);
		encoder.setTraveledDistance();
		} 
	);

	rpio.open(pinB, rpio.INPUT, rpio.PULL_DOWN);
	rpio.poll(pinB, function countTickPinB(pinB){
		//Returns either 0 for LOW or 1 for HIGH
		var statePinA = rpio.read(encoder.pinA);
		var statePinB = rpio.read(pinB);

		// look for a low-to-high on channel B
		if (statePinB == 1){
			// check channel A to see which way encoder is turning
			if (statePinA == 1) {
				encoder.nbTicks = encoder.nbTicks - 1; //CW
			}
			else {
				encoder.nbTicks = encoder.nbTicks + 1; //CCW
			}
		}
		// Look for a high-to-low on channel B
		else {
			// check channel B to see which way encoder is turning
			if (statePinA == 0) {
				encoder.nbTicks = encoder.nbTicks - 1; //CW
			}
			else {
				encoder.nbTicks = encoder.nbTicks + 1; //CCW
			}
		}
		console.log(name, " ", encoder.nbTicks);
		encoder.setTraveledDistance();
		} );
}
```

AntonPieper, 2023-03-14:
Couldn't you have bound the function `countTickPinA` to `this`?
```ts
rpio.poll(pinA, this.countTickPinA.bind(this));
```