jperkin/node-rpio #71: Error: Module version mismatch. Expected 46, got 57.

candelo, 2018-04-23:
Error: Module version mismatch. Expected 46, got 57.
    at Error (native)
    at Object.Module._extensions..node (module.js:434:18)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at bindings (/home/pi/LedTest/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/pi/LedTest/lib/rpio.js:17:34)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)

pi@raspberrypi:~/LedTest/examples $ node -v
v4.8.2

arovillard, 2018-10-27:
you are most likely running a node version lower than 8. if doing `sudo node -v` returns a version lower than 8 then you need to upgrade your node version. I had this issue and my system was running v 4.x.x. after updating it it worked as expected.

jperkin, 2018-11-29:
Right, you need to use the same version of node that you built with.  If you upgrade node then you need to rebuild.

closed 2018-11-29 by jperkin