jperkin/node-rpio #72: Installation Troubles

shard87, 2018-06-06:
Can anybody help with this... I'm literally getting stuck at the start and its no doubt something obvious I've missed!

Any time I type _npm install rpio_ I get this error:

_npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/rpio failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org:443_
_npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation._
_npm WARN checkPermissions Missing write access to /home/pi/node_modules/rpio_
_npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'_
_npm WARN pi No description_
_npm WARN pi No repository field._
_npm WARN pi No README data_
_npm WARN pi No license field._

_npm ERR! path /home/pi/node_modules/rpio_
_npm ERR! code EACCES_
_npm ERR! errno -13_
_npm ERR! syscall access_
_npm ERR! Error: EACCES: permission denied, access '/home/pi/node_modules/rpio'_
_npm ERR!  { Error: EACCES: permission denied, access '/home/pi/node_modules/rpio'_
_npm ERR!   stack: 'Error: EACCES: permission denied, access \'/home/pi/node_modules/rpio\'',_
_npm ERR!   errno: -13,_
_npm ERR!   code: 'EACCES',_
_npm ERR!   syscall: 'access',_
_npm ERR!   path: '/home/pi/node_modules/rpio' }_
_npm ERR!_
_npm ERR! Please try running this command again as root/Administrator._

_npm ERR! A complete log of this run can be found in:_
_npm ERR!     /home/pi/.npm/_logs/2018-06-06T07_55_14_019Z-debug.log_

If I run as SUDO I get this:

_npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/rpio failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org:443_
_npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation._

_> rpio@0.9.22 install /home/pi/node_modules/rpio_
_> node-gyp rebuild_

_gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/8.11.2"_
_gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/rpio/.node-gyp"_

The last 2 lines will keep repeating themselves until I stop it.

Any explanations would be very appreciated!

WeeMan1990, 2018-07-16:
I got some issues with this plugin as well, yours should however be easily fix by : 
sudo npm install --unsafe-perm rpio

shard87, 2018-07-16:
Thanks WeeMan,

I tried it but still had no luck:

pi@raspberrypi:~ $ sudo npm install --unsafe-perm rpio

> rpio@0.9.22 install /home/pi/node_modules/rpio
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/rpio/build'
  CC(target) Release/obj.target/rpio/src/bcm2835.o
  CXX(target) Release/obj.target/rpio/src/rpio.o
  SOLINK_MODULE(target) Release/obj.target/rpio.node
  COPY Release/rpio.node
make: Leaving directory '/home/pi/node_modules/rpio/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ rpio@0.9.22
added 3 packages in 17.435s

I assume it’s something ridiculous I’ve missed?

WeeMan1990, 2018-07-18:
what if you try § sudo npm install -g --unsafe-perm --verbose rpio?

WeeMan1990, 2018-07-18:
and also if you havn't done it already, check that you are a member of the GPIO group, 
§   groups ${USER} 
and if the return doesn't show gpio then run 
§ sudo usermod -a -g gpio pi
And ofcourse in the above ''pi'' is the username as standard, if you have changed your username you'll have to replace ''pi'' with the real

AlexSSD7, 2018-11-14:
--unsafe-perm helped me! Thanks!

jperkin, 2018-11-29:
This appears to be resolved now?  Sorry, I can't do much about NPM's terrible UNIX permissions handling unfortunately :(  The only thing I can do is highly highly highly recommend never using "-g" to install, it causes nothing but problems and makes either the operating system packagers or the module owners look bad.

closed 2018-11-29 by jperkin