jperkin/node-rpio #74: Crashing Raspberry Pi when trying to connect to input pin

adamreisnz, 2018-06-13:
Have you had any other reports of this? Everything used to work fine, but now whenever I try to connect to an input pin, the Raspberry Pi outright crashes/freezes and I have to manually reboot it.

Any idea's what could be happening? Any underlying packages that may have been updated and are causing failures with rpio?

This is with rpio version 0.9.22 and Node 8.10.0 as well as 8.11.3.

adamreisnz, 2018-06-13:
After some further analysis, I've managed to create a test script that reproduces the issue. However, the problem is that it doesn't reproduce it consistently.

```
const rpio = require('rpio');
const {INPUT, OUTPUT, HIGH, LOW, PULL_UP} = rpio;
const pin = 29;

console.log('Opening pin');
rpio.open(pin, INPUT, PULL_UP);
console.log('Pin opened');

console.log('Reading state');
const state = rpio.read(pin);
console.log(`Pin state is ${state}`);

console.log('Setting up polling');
rpio.poll(pin, pin => {
  console.log('In poll cycle');
  console.log('Reading state');
  const state = rpio.read(pin) || LOW
  console.log(`Pin state is ${state}`);
});
```

As soon as it hits the line with `rpio.poll` it crashes. The last console output is `Setting up polling`.

adamreisnz, 2018-06-13:
I've also tried with different pin numbers, and with calling the poll function after a timeout, all to no avail.

adamreisnz, 2018-06-13:
After running a few more tests, it appears the culprit is the pull up statement. After removing this, the device no longer seems to crash when reading/polling pin input.

It appears this issue is related https://github.com/raspberrypi/linux/issues/2550

ASwingler, 2018-06-14:
I've been battling lockups for the past couple of months. I think it's the same issue you've reported - it started happening after a kernel update (sorry - I didn't record the kernel version). This code consistently (but not always) causes a crash:

`var rpio = require('rpio');

var led = 17;
var btn = 2;

rpio.init({mapping: 'gpio'});
rpio.open(btn, rpio.INPUT, rpio.PULL_UP);
rpio.open(led, rpio.OUTPUT, rpio.HIGH);
rpio.poll(btn, pollcb);

function pollcb(pin)
{
        var state = rpio.read(pin) ? 'released' : 'pressed' ;
        rpio.write(led,rpio.read(pin));
        
}`

I've tested this on 3B and 3B+ hardware with the same result.

I noticed a few things:
1. If I run this code and then keeping pushing the button fast at the same time as repeatedly pressing the "enter" key on the console keyboard I can get it to happen.
2. The problem tends to manifest the first time I run node after a boot. If I stop and restart node I don't tend to see it.
3. If I use a different library (inout) I don't see this behavior. This isn't a solution for me however since other libraries don't have the functionality of rpio.
4. The OS does appear to keep running (top -id 0.25 shows ongoing activity). I appear to lose USB and network access however.
5. Prior to the kernel update (sometime in March I think) the code was running fine and was stable.
6. The problem appears whether I use /dev/gpiomem or /dev/mem*
This behavior exists on multiple devices. I've used multiple SDCards and built clean from the latest  Stretch Lite and the behavior is consistent. I see the problem on node 9.5 and 10.3.

Until your post today I wasn't aware that anyone else was having this problem, but it's a huge issue for me.

adamreisnz, 2018-06-14:
Yep, sounds like the same thing with the same symptoms, in particular number 2.

I'm creating a new image now as we speak using an older kernel version (4.9, from http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/) and once I have that up and running I will report back.

ASwingler, 2018-06-14:
Great. Let me know if you need me to test anything.

adamreisnz, 2018-06-14:
I can confirm that everything works as expected on 4.9. So the culprit was definitely the linked issue with the 4.14 kernel. Will close this one.

closed 2018-06-14 by adamreisnz

shiny, 2018-06-22:
same problem,  my script:
```javascript
const rpio = require('rpio');

const pin = 11;
rpio.open(pin, rpio.INPUT);

rpio.poll(pin, (pin) => {
  console.log('Button event on pin %d, is now %d', pin, rpio.read(pin));
});
```

My kernel is 4.14.34-v7+.

By the way,  this python script works well:
```python
try:
    import RPi.GPIO as GPIO
except RuntimeError:
    print("error")
        
import time
GPIO.setmode(GPIO.BOARD)
channel = 11
GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
cur = False
while True:
    tmp = GPIO.input(channel)
    if tmp == cur:
        time.sleep(0.05)
    else:
        print tmp
        cur = tmp
```

adamreisnz, 2018-06-23:
Did you check https://github.com/raspberrypi/linux/issues/2550?

zoolyka, 2020-04-19:
It still hangs with the recent (4.19.x) kernel. Any idea what causes it? So I can submit a ticket to raspbian developers. The latest working version is 4.9.80 _(can install it with: `sudo rpi-update 5c80565c5c0c7f820258c792a98b56f22db2dd03` - just as a note)_.

jperkin, 2020-04-19:
Did you enable `dtoverlay=gpio-no-irq`?

zoolyka, 2020-04-19:
Can confirm that `dtoverlay=gpio-no-irq` solves the hangups with recent kernels, thanks a lot!