jperkin/node-rpio #76: Importing rpio foe ES6

engineer-only, 2018-09-10:
I am trying to use rpio for my react APP that uses ES6 syntax. Normally I can use the write and read methods when I am using node and
`var rpio = require('rpio');`

However, when I use 
`import rpio from 'rpio';`

I catch an error that: 

Uncaught TypeError: exists is not a function
    at Function.getRoot (bindings.js:158)
    at bindings (bindings.js:60)
    at Object.<anonymous> (rpio.js:17)
    at Object../node_modules/rpio/lib/rpio.js (rpio.js:787)
    at __webpack_require__ (bootstrap 9ea40ba0fa93ccf9cb66:678)
    at fn (bootstrap 9ea40ba0fa93ccf9cb66:88)
    at Object../src/Services/switchController.js (switchController.js:1)
    at __webpack_require__ (bootstrap 9ea40ba0fa93ccf9cb66:678)
    

How can I fix this issue?

jperkin, 2018-09-10:
This sounds a lot like #53 with exactly the same error message.  Is this another environment where you do not have access to the file system?  I'm not familiar with ES6 or React, but `exists()` is a basic requirement and if they don't support it then this isn't going to work, sorry.

jaqua, 2018-10-08:
You have to use rpio on the server, not on the browser as a react app. Therefore you don't use ES6. It doesn't make sense to call rpio on the client side as you want to access the rpi pins.

engineer-only, 2018-10-11:
@jaqua Actually my React App is working in the local only, meaning that nobody can reach my React App except the machine that my node server works on. I wanted to remove the node server layer and make everything work with React. So that's why I opened this issue.But I guess there is no other solution rather then using NodeJS.

jaqua, 2018-10-11:
@engineer-only Even if you run everything local only, there is a client and a server side. Your react app is working on client side. Therefore it never gets access to the pins. You need a kind of API which is running on server side. There you get access to the pins. It doesn't matter if you use ES6 or not, therefore it is not a problem with `import...` vs `require()`

recallwolf, 2018-10-16:
You can use 'axios' to send the message to nodejs(maybe python or java) to tell the backend how to handle gpio.
Just like the jaqua said.

jolic, 2018-11-01:
```npm i --save-dev babel-cli babel-preset-env```
and create the file ```.babelrc``` with this content:

```
{
  "presets": [
    ["env", {
      "modules": false,
      "targets": {
        "node": "current"
      }
    }]
  ]
}
```

jperkin, 2018-11-29:
Closing this one out as this module is never going to be able to support being a client-side app.

closed 2018-11-29 by jperkin