jperkin/node-rpio #79: Usable for docker container?

jaqua, 2018-10-06:
I'm running some applications on my RPI via docker container.
Is it possible to use node-rpio inside a docker container? Do I have to make some directories accessable or something like this?

chriskinsman, 2020-07-28:
@jaqua What base image are you using?  I am hitting issues with missing /sys/device-tree on alpine

bdsoha, 2021-09-16:
I am having an issue trying to run this on K8S.

In order to control the *GPIO* pins, I mapped the following host directories:
- `/sys/class/gpio`
- `/sys/devices/platform/soc`
- `/dev/gpiomem`
- `/proc/device-tree`
- `/sys/firmware/devicetree/base`

There is no issue running the following from within the container:
```bash
$ echo 21 >/sys/class/gpio/export
$ echo out >/sys/class/gpio/gpio21/direction
$ echo 1 >/sys/class/gpio/gpio21/value
$ echo 0 >/sys/class/gpio/gpio21/value
```