jperkin/node-rpio #80: bout the pwm, servo don't work correctly

recallwolf, 2018-10-17:
Sorry to interrupt you. This has been troubling for a few days.

servo(sg90) need 50hz(20ms) time base pulse.
0.5ms-------------0°
1.0ms------------45°
1.5ms------------90°
2.0ms-----------135°
2.5ms-----------180°

According to this formula:

pwmFrequency in Hz = 19.2e6 Hz / pwmClock / pwmRange

I tried pwmClock(64/128/256.....) and pwmRange(6000/3000/1500) but only 64/6000 128/3000 works.
And work uncorrectly.

Here is the code 
``var rpio = require('rpio');
rpio.init({gpiomem: false});
rpio.open(33, rpio.PWM);
rpio.pwmSetClockDivider(128);
rpio.pwmSetRange(33, 3000);
rpio.pwmSetData(33, 150);
``

``var rpio = require('rpio');
rpio.init({gpiomem: false});
rpio.open(33, rpio.PWM);
rpio.pwmSetClockDivider(64);
rpio.pwmSetRange(33, 6000);
rpio.pwmSetData(33, 300);
``
The first one whatever I change the data to 150/225/300/375, It only turn to 90°
The second one change the data, It only turn to 0°

Is there any problem with the code?
Hope someone can point out.
thank you very much!

yutou527, 2019-11-01:
i use the RPi.GPIO succed drive the servo, and meet the same problem.
i think that you should calculate the interval (T) with  19.2e6 Hz / pwmClock / pwmRange 
and then , use range * 2.5ms/T to get real data range.
for example 
pwmClock = 128 
pwmRange = 1000
19200000 / 128 / 2000 = 75Hz
1000ms/75= 13.33ms
2.5/13.33 = 0.1875
so your data range should between 0-187.5 (1000*0.1875)
i guess so. but not very sure.

yutou527, 2019-11-01:
@recallwolf

GitMcClintock, 2021-04-20:
I got this going using nodejs-rpio as my GPIO library, and the following settings:

Clock divider = 256
Range = 1500

These values give a 20 ms period (50 Hz rate) on the PWM output.

Then a value of 38 gives a pulse with of 0.5 ms,
and:
- 113 gives 0.5 ms
- 188 gives 2.5 ms

My servo required this 0.5 to 2.5 ms range for a full 180 degrees of rotation.  Yours may be different.

Tested on my Raspberry Pi 3A+, using pin 33.