jperkin/node-rpio #81: Polling does not work

Mrtenz, 2018-10-26:
I'm trying to use the `poll` function but it does not seem to work. Manually polling with `setInterval` does work.

Here is the code that's not working:
```typescript
import { init, open, poll, INPUT, PULL_UP } from 'rpio';

// I have to use `gpiomem: false` because I'm using SPI functions
init({
  gpiomem: false,
  mapping: 'gpio'
});

open(21, INPUT, PULL_UP);
poll(21, () => {
  console.log('Foo'); // Never called
});
```

While using `setInterval` and `read` does work:

```typescript
import { init, open, poll, read, INPUT, LOW, PULL_UP } from 'rpio';

// I have to use `gpiomem: false` because I'm using SPI functions
init({
  gpiomem: false,
  mapping: 'gpio'
});

open(21, INPUT, PULL_UP);
setInterval(() => {
  if (read(21) === LOW) {
    console.log('Foo'); // Called when button is pressed
  }
}, 10);
```

Am I missing something here or is this a bug in `node-rpio`?

jperkin, 2018-11-29:
I wonder if it's just that polling isn't supported for SPI.  Out of interest, to rule out any hardware or kernel issues, if you just use regular GPIO does it work?

If it's that SPI and polling are incompatible then I'm not sure there is a fix.

olecom, 2019-02-12:
As far as I can see, on SPI master cannot wait for a message to read from a slave (e.g. using `poll`/epoll/select). Because master hardware must select slave via PIN and then start clocking, after what slave will provide bits serially.

OTOH events can be seen by SPI slaves, when a message from master arrives.

> I'm trying to use the `poll`

What SPI master/slave setup is there and who is reading?