jperkin/node-rpio #82: V8 Deprecation Warnings

intervalia, 2018-11-11:
I am getting the following V8 Deprecation Warnings on MAC and my PI.

**I am using Node version 10.12.0 on my MAC and 10.13.0 my PI 3 b+**

Here is the G++ info from my PI:

```
g++ (Raspbian 4.9.2-10+deb8u1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Here is the output from my `npm install`:

```
../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)’:
../../nan/nan_implementation_12_inl.h:340:37: warning: ‘static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:5052): Use Isolate* version [-Wdeprecated-declarations]
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
In file included from ../src/rpio.cc:17:0:
../../nan/nan.h: In constructor ‘Nan::Utf8String::Utf8String(v8::Local<v8::Value>)’:
../../nan/nan.h:1066:53: warning: ‘v8::Local<v8::String> v8::Value::ToString() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:10040): Use maybe version [-Wdeprecated-declarations]
       v8::Local<v8::String> string = from->ToString();
                                                     ^
../../nan/nan.h:1080:74: warning: ‘int v8::String::WriteUtf8(char*, int, int*, int) const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2658): Use Isolate* version [-Wdeprecated-declarations]
         length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
                                                                          ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_function(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:34:28: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
      (info[1]->NumberValue() > 7))
                            ^
../src/rpio.cc:37:41: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  bcm2835_gpio_fsel(info[0]->NumberValue(), info[1]->NumberValue());
                                         ^
../src/rpio.cc:37:65: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  bcm2835_gpio_fsel(info[0]->NumberValue(), info[1]->NumberValue());
                                                                 ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_read(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:48:66: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  info.GetReturnValue().Set(bcm2835_gpio_lev(info[0]->NumberValue()));
                                                                  ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_readbuf(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:62:45: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
  buf = node::Buffer::Data(info[1]->ToObject());
                                             ^
../src/rpio.cc:64:39: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  for (i = 0; i < info[2]->NumberValue(); i++)
                                       ^
../src/rpio.cc:65:50: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
   buf[i] = bcm2835_gpio_lev(info[0]->NumberValue());
                                                  ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_write(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:75:42: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  bcm2835_gpio_write(info[0]->NumberValue(), info[1]->NumberValue());
                                          ^
../src/rpio.cc:75:66: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  bcm2835_gpio_write(info[0]->NumberValue(), info[1]->NumberValue());
                                                                  ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_writebuf(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:89:45: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
  buf = node::Buffer::Data(info[1]->ToObject());
                                             ^
../src/rpio.cc:91:39: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  for (i = 0; i < info[2]->NumberValue(); i++)
                                       ^
../src/rpio.cc:92:43: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
   bcm2835_gpio_write(info[0]->NumberValue(), buf[i]);
                                           ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE gpio_pad_read(Nan::NAN_METHOD_ARGS_TYPE)’:
[... 140 lines cut ...]
           node::Buffer::Data(info[1]->ToObject()),
                                                ^
../src/rpio.cc:357:32: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
           info[2]->NumberValue());
                                ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE spi_write(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:367:59: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
  bcm2835_spi_writenb(node::Buffer::Data(info[0]->ToObject()),
                                                           ^
../src/rpio.cc:368:29: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
        info[1]->NumberValue());
                             ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE rpio_init(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:384:41: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  if (!bcm2835_init(info[0]->NumberValue()))
                                         ^
../src/rpio.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE rpio_usleep(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rpio.cc:401:30: warning: ‘double v8::Value::NumberValue() const’ is deprecated (declared at /home/pi/.node-gyp/10.13.0/include/node/v8.h:2475): Use maybe version [-Wdeprecated-declarations]
  usleep(info[0]->NumberValue());
                              ^
  SOLINK_MODULE(target) Release/obj.target/rpio.node
  COPY Release/rpio.node
make: Leaving directory '/home/pi/projects/lights/node_modules/rpio/build'
npm WARN lights@1.0.0 No repository field.
```

jperkin, 2018-11-29:
Thanks, I'll take a look at this.  Hopefully NAN provides a way to abstract this so that it will continue to work with older V8s.

ewertoncostadelle, 2018-12-17:
> Thanks, I'll take a look at this. Hopefully NAN provides a way to abstract this so that it will continue to work with older V8s.

which V8's version are you using?

JimSchofield, 2019-05-05:
I had the same issues with node 12, but rolled back to node 8.1 with NVM and install worked fine.