jperkin/node-rpio #90: Can't install on Windows

Cyberuben, 2018-11-29:
I'm not able to install this repo on Windows. I'm running node `v10.12.0`, npm `v10.12.0`, node-gyp `v3.8.0`.

The errors I'm prompted with are the following:

```
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  bcm2835.c
c:\users\ruben\projects\rpi\node_modules\rpio\src\bcm2835.c(16): fatal error C1083: Cannot open include file: 'sys/mman.h': No such file or directory [C:\Users\Ruben\Projects\RPI\node_modules\rpio\build\rpio.vcxproj]
```

I can't seem to find any issues regarding this on Windows, so I'm not sure if what I'm doing is supported at all. I know there's a mock option, but that's of no use if I can't get it to install.

jperkin, 2018-11-29:
Yeh this module doesn't support Windows I am afraid.  It is currently tied heavily to the Linux kernel.  Cross platform support is something I am passionate about and would be fantastic to have, but would be a lot of work even for a different POSIX platform.  Windows would be completely different, and I have no Windows kernel programming experience.  I'll close this one out for now, but if someone wanted to implement Windows support via a pull request at some point I would be very interested!  Thanks.

closed 2018-11-29 by jperkin

Cyberuben, 2018-11-29:
Thanks for the quick response. It should work on any Linux environment? If so, I'll try developing this on a VM.

jperkin, 2018-11-29:
Yes, it should, if not let me know!