jperkin/node-rpio #97: "Hardware auto-detect failed" on RPi 3B+ with ARM64 Ubuntu

azazdeaz, 2019-04-21:
Hi there,

I'm getting the following warning on RPi 3B+:
`WARNING: Hardware auto-detect failed, running in raspi-3 mock mode`

I'm using the preinstalled [Ubuntu 18.04.2 ARM64 for RPi image](http://cdimage.ubuntu.com/releases/bionic/release/)

<details><summary>$ cat /proc/cpuinfo</summary>
<pre>
cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
</pre>
</details>

<details><summary>$ sudo lshw</summary>
<pre>
ubuntu                      
    description: Computer
    product: Raspberry Pi 3 Model B Plus Rev 1.3
    serial: 0000000023641601
    width: 64 bits
    capabilities: smp cp15_barrier setend swp
  *-core
       description: Motherboard
       physical id: 0
     *-cpu:0
          description: CPU
          product: cpu
          physical id: 0
          bus info: cpu@0
          size: 1400MHz
          capacity: 1400MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:1
          description: CPU
          product: cpu
          physical id: 1
          bus info: cpu@1
          size: 1400MHz
          capacity: 1400MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:2
          description: CPU
          product: cpu
          physical id: 2
          bus info: cpu@2
          size: 1400MHz
          capacity: 1400MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:3
          description: CPU
          product: cpu
          physical id: 3
          bus info: cpu@3
          size: 1400MHz
          capacity: 1400MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-memory
          description: System memory
          physical id: 4
          size: 912MiB
  *-usbhost
       product: DWC OTG Controller
       vendor: Linux 4.15.0-1033-raspi2 dwc_otg_hcd
       physical id: 1
       bus info: usb@1
       logical name: usb1
       version: 4.15
       capabilities: usb-2.00
       configuration: driver=hub slots=1 speed=480Mbit/s
     *-usb
          description: USB hub
          product: USB 2.0 Hub
          vendor: Standard Microsystems Corp.
          physical id: 1
          bus info: usb@1:1
          version: b.b3
          capabilities: usb-2.00
          configuration: driver=hub maxpower=2mA slots=4 speed=480Mbit/s
        *-usb
             description: USB hub
             product: USB 2.0 Hub
             vendor: Standard Microsystems Corp.
             physical id: 1
             bus info: usb@1:1.1
             version: b.b3
             capabilities: usb-2.00
             configuration: driver=hub maxpower=2mA slots=3 speed=480Mbit/s
           *-usb
                description: Generic USB device
                vendor: Standard Microsystems Corp.
                physical id: 1
                bus info: usb@1:1.1.1
                version: 3.00
                capabilities: usb-2.10
                configuration: driver=lan78xx maxpower=2mA speed=480Mbit/s
  *-network:0
       description: Ethernet interface
       physical id: 2
       logical name: docker0
       serial: 02:42:b2:df:9d:c1
       capabilities: ethernet physical
       configuration: broadcast=yes driver=bridge driverversion=2.3 firmware=N/A ip=172.17.0.1 link=no multicast=yes
  *-network:1 DISABLED
       description: Wireless interface
       physical id: 3
       logical name: wlan0
       serial: b8:27:eb:31:43:54
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=brcmfmac driverversion=7.45.18 firmware=01-6a2c8ad4 multicast=yes wireless=IEEE 802.11
  *-network:2
       description: Ethernet interface
       physical id: 4
       logical name: eth0
       serial: b8:27:eb:64:16:01
       size: 1Gbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=lan78xx driverversion=1.0.6 duplex=full ip=10.42.0.45 link=yes multicast=yes port=MII speed=1Gbit/s
</pre>
</details>

RPi.GPIO and [gpiozero](https://gpiozero.readthedocs.io/en/stable/) are failing with similar errors but i'm not sure how should i continue debugging :confused: 

`rpio@1.0.13` `node v10.15.3`

jperkin, 2019-04-21:
This is the same as #29 I think, I'm not aware of any 64-bit support in the upstream bcm2835 library yet.

closed 2019-07-13 by azazdeaz

frederic-bonjour, 2019-10-14:
Hi :)
I'm facing the same issue now.
@azazdeaz Did you finally get things working on yout RPi 3B+?
Thanks for your answers!

jperkin, 2019-10-14:
Did you try the latest version? It should detect based on device tree.