jperkin/node-rpio #101: rpio not compiling on node 12

KryQ, 2019-07-31:
Hi so upon installing package from npm rpio simply isn't compilling due deprecation errors.

Did somebody resolver this issue?

`> rpio@1.0.13 install /home/pi/Servers/radarhead_lst/node_modules/rpio
> node-gyp rebuild

make: Entering directory '/home/pi/Servers/radarhead_lst/node_modules/rpio/build'
  CC(target) Release/obj.target/rpio/src/bcm2835.o
  CXX(target) Release/obj.target/rpio/src/rpio.o
In file included from ../node_modules/nan/nan_converters.h:67,
                 from ../node_modules/nan/nan.h:221,
                 from ../src/rpio.cc:17:
../node_modules/nan/nan_converters_43_inl.h: In static member function 'static Nan::imp::ToFactoryBase<v8::Boolean>::return_t Nan::imp::ToFactory<v8::Boolean>::convert(v8::Local<v8::Value>)':
../node_modules/nan/nan_converters_43_inl.h:18:51: warning: 'v8::MaybeLocal<v8::Boolean> v8::Value::ToBoolean(v8::Local<v8::Context>) const' is deprecated: ToBoolean can never throw. Use Local version. [-Wdeprecated-declarations]
       val->To ## TYPE(isolate->GetCurrentContext())                            \
                                                   ^
../node_modules/nan/nan_converters_43_inl.h:22:1: note: in expansion of macro 'X'
 X(Boolean)
 ^
In file included from /home/pi/.node-gyp/12.7.0/include/node/v8-internal.h:14,
                 from /home/pi/.node-gyp/12.7.0/include/node/v8.h:25,
                 from /home/pi/.node-gyp/12.7.0/include/node/node.h:63,
                 from ../node_modules/nan/nan.h:52,
                 from ../src/rpio.cc:17:
/home/pi/.node-gyp/12.7.0/include/node/v8.h:2531:59: note: declared here
                 V8_WARN_UNUSED_RESULT MaybeLocal<Boolean> ToBoolean(
                                                           ^~~~~~~~~
/home/pi/.node-gyp/12.7.0/include/node/v8config.h:311:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
In file included from ../node_modules/nan/nan_converters.h:67,
                 from ../node_modules/nan/nan.h:221,
                 from ../src/rpio.cc:17:
../node_modules/nan/nan_converters_43_inl.h: In static member function 'static Nan::imp::ValueFactoryBase<bool>::return_t Nan::imp::ToFactory<bool>::convert(v8::Local<v8::Value>)':
../node_modules/nan/nan_converters_43_inl.h:37:57: warning: 'v8::Maybe<bool> v8::Value::BooleanValue(v8::Local<v8::Context>) const' is deprecated: BooleanValue can never throw. Use Isolate version. [-Wdeprecated-declarations]
   return val->NAME ## Value(isolate->GetCurrentContext());                     \
                                                         ^
../node_modules/nan/nan_converters_43_inl.h:40:1: note: in expansion of macro 'X'
 X(bool, Boolean)
 ^
In file included from /home/pi/.node-gyp/12.7.0/include/node/v8-internal.h:14,
                 from /home/pi/.node-gyp/12.7.0/include/node/v8.h:25,
                 from /home/pi/.node-gyp/12.7.0/include/node/node.h:63,
                 from ../node_modules/nan/nan.h:52,
                 from ../src/rpio.cc:17:
/home/pi/.node-gyp/12.7.0/include/node/v8.h:2569:51: note: declared here
                 V8_WARN_UNUSED_RESULT Maybe<bool> BooleanValue(
                                                   ^~~~~~~~~~~~
/home/pi/.node-gyp/12.7.0/include/node/v8config.h:311:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
In file included from ../node_modules/nan/nan_new.h:189,
                 from ../node_modules/nan/nan.h:222,
                 from ../src/rpio.cc:17:
../node_modules/nan/nan_implementation_12_inl.h: In static member function 'static Nan::imp::FactoryBase<v8::Function>::return_t Nan::imp::Factory<v8::Function>::New(Nan::FunctionCallback, v8::Local<v8::Value>)':
../node_modules/nan/nan_implementation_12_inl.h:105:32: error: no matching function for call to 'v8::Function::New(v8::Isolate*&, void (&)(const v8::FunctionCallbackInfo<v8::Value>&), v8::Local<v8::Object>&)'
                           , obj));
                                ^
In file included from /home/pi/.node-gyp/12.7.0/include/node/node.h:63,
                 from ../node_modules/nan/nan.h:52,
                 from ../src/rpio.cc:17:
/home/pi/.node-gyp/12.7.0/include/node/v8.h:4132:31: note: candidate: 'static v8::MaybeLocal<v8::Function> v8::Function::New(v8::Local<v8::Context>, v8::FunctionCallback, v8::Local<v8::Value>, int, v8::ConstructorBehavior, v8::SideEffectType)'
   static MaybeLocal<Function> New(
                               ^~~
/home/pi/.node-gyp/12.7.0/include/node/v8.h:4132:31: note:   no known conversion for argument 1 from 'v8::Isolate*' to 'v8::Local<v8::Context>'
In file included from ../src/rpio.cc:17:
../node_modules/nan/nan.h: In constructor 'Nan::Utf8String::Utf8String(v8::Local<v8::Value>)':
../node_modules/nan/nan.h:1064:78: warning: 'v8::Local<v8::String> v8::Value::ToString(v8::Isolate*) const' is deprecated: Use maybe version [-Wdeprecated-declarations]
       v8::Local<v8::String> string = from->ToString(v8::Isolate::GetCurrent());
[... 898 lines cut ...]
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:266:23)
gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.58-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Servers/radarhead_lst/node_modules/rpio
gyp ERR! node -v v12.7.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN uart_listener@1.0.0 No description
npm WARN uart_listener@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rpio@1.0.13 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rpio@1.0.13 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-31T07_49_40_878Z-debug.log
`

ARitz-Cracker, 2019-08-07:
Please wrap log in multi-line code tags next time like this: \`\`\`text here\`\`\`.
Anyway, yeah. This module is currently incompatible with node v12 due to the desire by the author to keep this thing compatible with old versions of node (see #82).

Though since [nan](https://www.npmjs.com/package/nan) was updated 3 months ago, and this module was updated 6 months ago, perhaps some tweaking after updating the dependencies will help, though I personally don't know much about creating binary modules for node so I can't be sure. @jperkin or someone familiar with the inner workings of how this works would have to take a look.

Anyway, right now I'm using Node 10 with this module since it's currently in LTS.

KryQ, 2019-09-13:
> same problem here

check my pull request i fixed it there

jperkin, 2019-10-10:
Node v12.x is now supported as of 1.2.x.

closed 2019-10-10 by jperkin