jperkin/node-rpio #104: Raspberry PI 4 support

Serveurperso, 2019-08-12:
bcm2835.c look too old for rpi4 I try to upgrade from https://www.airspayce.com/mikem/bcm2835/ & rebuild but same problem...

KryQ, 2019-08-21:
Check my pull request i updated this library to rpi4 and node 12. Feedback appreciated

Serveurperso, 2019-08-21:
Thanks a lot for your work

jperkin, 2019-10-02:
I just pushed v1.1.0 which updates bcm2835 to the latest which allegedly supports RPi 4, as well as adding a board id.  Would be curious to know if this now works for you.

jacobq, 2019-10-09:
@jperkin I just tried using this library for the first time yesterday, and I was able to read and write GPIO pins on an RPi 4 running that latest raspbian. I had to use Node v10.x instead of 12.x though (see https://github.com/jperkin/node-rpio/issues/101)

jperkin, 2019-10-09:
Great! I'll be addressing the node 12.x issues in a separate fix.

closed 2019-10-09 by jperkin