jperkin/node-rpio #105: WARNING: Hardware auto-detect failed, running in raspi-3 mock mode

dauledk, 2019-10-11:
I am trying to run `node-rpio` on the raspberry pi 4, with the [64-bit balenaOS](https://www.balena.io/blog/balena-releases-first-fully-functional-64-bit-os-for-the-raspberry-pi-4/).

I am running the docker image `balenalib/raspberrypi4-64-node:12`, and everything builds ok. But when I start the Node process using rpio, I get this warning: 

```
WARNING: Hardware auto-detect failed, running in raspi-3 mock mode
```

Do you have any ideas of how to fix this?

jperkin, 2019-10-11:
The code which tries to detect the hardware is [here](https://github.com/jperkin/node-rpio/blob/master/lib/rpio.js#L234-L296), specifically it looks at the output of `/proc/cpuinfo` and attempts to parse the line starting "Revision".

In your case it isn't finding a match.  Could you paste what that line contains on your OS?

dauledk, 2019-10-11:
Ahh ok - would love to get involved, so thanks for pointing out the file detecting the hardware. Here is the output:

```
cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3
```

jperkin, 2019-10-11:
Ok, so your OS differs from all the others that this module has been tested on so far, and doesn't include the board revision. Could you find out where it stores this information instead?

jperkin, 2019-10-11:
Actually according to [this](https://forums.balena.io/t/rpi4-proc-cpuinfo-missing-hardware-and-model-name/29080/4) it's stored in different files - could you output their contents?

dauledk, 2019-10-11:
The Balena OS and developer experience is really outstanding, makes it so easy to deploy and maintain code on raspberry pi's. So would be awesome if we could make it work with `node-rpio`. Btw, thanks for an awesome repo 🚀 Here is the output of the files:

**/sys/firmware/devicetree/base/serial-number**
```
100000001cbcbcaf
```
**/proc/device-tree/serial-number**
```
100000001cbcbcaf
```
**/sys/firmware/devicetree/base/model**
```
Raspberry Pi 4 Model B Rev 1.1
```
**/proc/device-tree/model**
```
Raspberry Pi 4 Model B Rev 1.1
```

jperkin, 2019-10-11:
Hmm, are there any other files in those directories that might be more useful? The serial number is going to differ between devices (I would imagine) and the model isn't great for parsing.

dauledk, 2019-10-11:
I am not that much into Linux - but maybe this [comment could help](https://github.com/raspberrypi/linux/issues/2110#issuecomment-467827060)?

jperkin, 2019-10-11:
I just pushed v1.2.3 which should fix this, can you try that?

dauledk, 2019-10-11:
@jperkin - you are fast 🤯

v1.2.3 seems to fix the issue - I am not seeing `WARNING: Hardware auto-detect failed, running in raspi-3 mock mode`! Looking forward to starting the actual project later today, where I will try to implement the poll/callback example for adding buttons.

dauledk, 2019-10-12:
Tested with a button - and it worked on Raspberry pi 4. Thanks for the help!

closed 2019-10-12 by dauledk

bdsoha, 2021-09-14:
Not able to detect hardware from docker running on K3OS:

```
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : d03114
Serial          : 10000000f5333f5a
Model           : Raspberry Pi 4 Model B Rev 1.4
```

zumdar, 2022-03-14:
im also having this:

WARNING: Hardware auto-detect failed, running in raspi-3 mock mode problem 

:-(

ive tried everything in this thread. even tried the mounting a fakecpu info file but that didnt work. 

I also have Pi 4 Model B Rev 1.1