jperkin/node-rpio #108: Simple rpio i2c example doesn't release the line

alexandrbig, 2019-10-26:
Sensor: bmp180

```
$ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```

Model: RPI 2B+

rpio: 1.3.0 

Example is simple `i2c.js`:
```
var rpio = require('rpio');

rpio.i2cBegin();
rpio.i2cSetSlaveAddress(0x77);
rpio.i2cSetBaudRate(100000);

rpio.i2cEnd();
```

**Steps:**
`$ sudo i2cdetect -y 1`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77
```

`$ sudo node i2c.js`

After that the command `sudo i2cdetect -y 1` runs 10 times slower and shows no sensor:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

More over, I have another sensors which can be read with `rpio` without any issue, but after first read any other program **can't** get access to i2c bus.

For example:
After reboot:
```
$ sudo node ~/api/bmp180.js
{ pressure: 100472.20418543817, temperature: 21.7 }
```

```$ sudo node ~/api/i2c.js```

```
$ sudo node ~/api/bmp180.js
(node:4666) UnhandledPromiseRejectionWarning: Error: EIO, Input/output error
(node:4666) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:4666) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

bmp180.js:
```
const bmp180 = require('bmp180-sensor')

async function readBmp180() {
    const sensor =
    await bmp180({
        address: 0x77, mode: 3,
    })

    const data = await sensor.read()

    console.log(data)

    await sensor.close()
}
readBmp180();
```

PS: 
`dtoverlay=gpio-no-irq` 
`$ cat >/etc/udev/rules.d/20-gpiomem.rules <<EOF
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="gpio", MODE="0660"
EOF`
with and without it story is the same

Please let me know what is wrong and how to fix it.
Thank you.

jperkin, 2019-10-26:
Thanks for the report. I only have a couple of i2c devices which appear to work ok, but will try to see if I can figure out if there are any issues with init/deinit.

alexandrbig, 2019-10-26:
Thank you for the quick reply.
If there is any help needed from my side - just let me know.

PeterMaes, 2020-04-05:
same issue here !
I have a few devices on the I2C bus.
At first run of i2cdetect the RPi sees all devices at full speed.
After running the NodeJS application (RPIO LCD example), the display is refreshed,
but afterwards i2cdetect is very slow and it does'tn see the devices anymore.

$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: 20 21 -- -- -- -- -- 27 -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --

$ sudo node index.js (this is the LCD example)

$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

i2cdetect sees something at address 0x03 , but after the next run no devices are detected

jrsanchezalcala, 2020-08-13:
Happen the same as **PeterMaes** , after rpio.i2cBegin();

k3341095, 2023-01-17:
I have the same problem.. no fix, no solution.
Has anyone solved it?
node.js v18 lte/raspberry pi zero 2w