jperkin/node-rpio #109: dht11 don't work

fabioce, 2019-10-26:
I'm trying the dht11.js file with my raspberry pi3 and my sensor dht11.
It answer me this:
<Buffer 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 ... >
Read failed:
    chk = 0, chksum = NaN
    NaN NaN NaN NaN (0 = NaN)

I printed also the buffer to understand.

I have also a python script that work fine.
You can see the python script here:
https://github.com/adeept/Adeept_RFID_Learning_Kit_Python_Code_for_RPi/blob/master/15_dht11.py

there are some differences in the steps to start sensor to send data.
for example see here:

while GPIO.input(channel) == GPIO.LOW:
	continue

while GPIO.input(channel) == GPIO.HIGH:
	continue

while j < 40:
	k = 0
	while GPIO.input(channel) == GPIO.LOW:
		continue

	while GPIO.input(channel) == GPIO.HIGH:
		k += 1
		if k > 100:
			break

	if k < 8:
		data.append(0)
	else:
		data.append(1)

	j += 1

print "sensor is working."
print data

What do you think?
Thanks

jperkin, 2019-10-26:
There is a specific timing requirement in this script when initialising the device, according to the datasheet we need to set low for 18us before setting high, but this is difficult to achieve from JavaScript with the function call overhead.  The python script is doing something similar, but is waiting for a different length of time (2us) compared to our 3us.

So, try changing the `rpio.msleep(3)` in dht11.js to some different values and see if that helps.

As for a more elegant solution, I should add a new API call which allows a more specific value to be set when cycling a pin.  I'll take a look at doing that, then if you manage to get this working we can verify that it improves things.

fabioce, 2019-10-26:
Thanks, but nothing.
I tryed rpio.msleep from 1 to 20 and the result is always the same
Read failed:

jperkin, 2019-10-26:
Ok, I'll try to find some time soon to dig out my hardware and try to verify.

jperkin, 2020-01-15:
This should work now with the example in v2.1.0, let me know if it doesn't.

closed 2020-01-15 by jperkin