jperkin/node-rpio #110: compiled against a different Node.js version using NODE_MODULE_VERSION 64

DonavanMartin, 2020-01-08:
Error: The module '.../node_modules/rpio/build/Release/rpio.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:718:18)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at bindings (.../node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (.../node_modules/rpio/lib/rpio.js:17:34)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)

DonavanMartin, 2020-01-08:
after trying `npm install `:

Error: The module '.../node_modules/rpio/build/Release/rpio.node'
was compiled against a different Node.js version using
**NODE_MODULE_VERSION 64**. This version of Node.js requires
**NODE_MODULE_VERSION 72**. Please try re-compiling or re-installing

closed 2020-01-08 by DonavanMartin

Yorams, 2020-01-08:
Maybe try to install a newer version of Node with this manual: https://github.com/nodesource/distributions

DonavanMartin, 2020-01-08:
Curiously I Solved it by running:
`npm install rpio`
`sudo yarn start`

node -v --> v12.14.0